tier4/AutowareArchitectureProposal.proj #40: please make sure tensorrt version is 7

sdu2011, 2020-06-19:
in PluginFactory.h
```
using nvinfer1::plugin::createPReLUPlugin;
```

however,in tensorrt7 header file NvinferPlugin.h,there is no createPReLUPlugin at all.

yukkysaito, 2020-06-19:
Thank you for trying.

Can you tell me the results of the error output?
And which ros package?

sdu2011, 2020-06-22:
> Thank you for trying.
> 
> Can you tell me the results of the error output?
```
In file included from /home/nv/tier4_trt_yolov3/src/tensorrt_yolo3/lib/include/TrtNet.h:34:0,
                 from /home/nv/tier4_trt_yolov3/src/tensorrt_yolo3/lib/src/TrtNet.cpp:24:
/home/nv/tier4_trt_yolov3/src/tensorrt_yolo3/lib/include/PluginFactory.h:44:25: error: ‘nvinfer1::plugin::createPReLUPlugin’ has not been declared
 using nvinfer1::plugin::createPReLUPlugin;
```
> And which ros package?
perception/object_recognition/detection/tensorrt_yolo3

sergiev, 2020-06-22:
@sdu2011 I don't think that 5th tensorrt is compatible with some perception libs
[image]
.

sdu2011, 2020-06-22:
> @sdu2011 I don't think that 5th tensorrt is compatible with some perception packages
> [image]
> .

it seems that the code is not tested well. maybe different package depends on different tensorrt version.  the api tensort_yolo3 package use exists in tensorrt5 header file.　and when i compile lidar_apollo_instance_segmentation with tensorrt7,it works fine.

yukkysaito, 2020-06-22:
thank you for your reply.

In my environment (TensorRT7), there seems to be a createPReLUPlugin.

```
yukihiro@yukihiro:/usr/include$ grep -r "createPReLUPlugin" /usr
NvInferPlugin.h:TRT_DEPRECATED_API INvPlugin* createPReLUPlugin(float negSlope);
TRT_DEPRECATED_API INvPlugin* createPReLUPlugin(const void* data, size_t length);
```

In our environment, we are able to build with TensorRT7, but I don't know why...
Can you run the following command and let me know the results?
`dpkg -l | grep nvinfer`

#### My Environment
```
yukihiro@yukihiro:~/workspace/AutowareArchitectureProposal$ dpkg -l | grep nvinfer
hi  libnvinfer-dev                               7.0.0-1+cuda10.2                                 amd64        TensorRT development libraries and headers
hi  libnvinfer-plugin-dev                        7.0.0-1+cuda10.2                                 amd64        TensorRT plugin libraries
hi  libnvinfer-plugin7                           7.0.0-1+cuda10.2                                 amd64        TensorRT plugin libraries
hi  libnvinfer7                                  7.0.0-1+cuda10.2                                 amd64        TensorRT runtime libraries
```

sergiev, 2020-06-22:
@sdu2011 just installed 7th TensorRT to check if TLR part works as intended:
```
dpkg -l | grep TensorRT
ii  graphsurgeon-tf                                             7.1.3-1+cuda10.2                                   amd64        GraphSurgeon for TensorRT package
ii  libnvinfer-bin                                              7.1.3-1+cuda10.2                                   amd64        TensorRT binaries
ii  libnvinfer-dev                                              7.1.3-1+cuda10.2                                   amd64        TensorRT development libraries and headers
ii  libnvinfer-doc                                              7.1.3-1+cuda10.2                                   all          TensorRT documentation
ii  libnvinfer-plugin-dev                                       7.1.3-1+cuda10.2                                   amd64        TensorRT plugin libraries
ii  libnvinfer-plugin7                                          7.1.3-1+cuda10.2                                   amd64        TensorRT plugin libraries
ii  libnvinfer-samples                                          7.1.3-1+cuda10.2                                   all          TensorRT samples
ii  libnvinfer5                                                 5.1.5-1+cuda10.0                                   amd64        TensorRT runtime libraries
ii  libnvinfer7                                                 7.1.3-1+cuda10.2                                   amd64        TensorRT runtime libraries
ii  libnvonnxparsers-dev                                        7.1.3-1+cuda10.2                                   amd64        TensorRT ONNX libraries
ii  libnvonnxparsers7                                           7.1.3-1+cuda10.2                                   amd64        TensorRT ONNX libraries
ii  libnvparsers-dev                                            7.1.3-1+cuda10.2                                   amd64        TensorRT parsers libraries
ii  libnvparsers7                                               7.1.3-1+cuda10.2                                   amd64        TensorRT parsers libraries
ii  python3-libnvinfer                                          7.1.3-1+cuda10.2                                   amd64        Python 3 bindings for TensorRT
ii  python3-libnvinfer-dev                                      7.1.3-1+cuda10.2                                   amd64        Python 3 development package for TensorRT
ii  tensorrt                                                    7.1.3.4-1+cuda10.2                                 amd64        Meta package of TensorRT
ii  uff-converter-tf                                            7.1.3-1+cuda10.2                                   amd64        UFF converter for TensorRT package
```
[image]
Though with warnings, it can be built.

sdu2011, 2020-06-23:
> @sdu2011
> 
> thank you for your reply.
> 
> In my environment (TensorRT7), there seems to be a createPReLUPlugin.
> 
> ```
> yukihiro@yukihiro:/usr/include$ grep -r "createPReLUPlugin" /usr
> NvInferPlugin.h:TRT_DEPRECATED_API INvPlugin* createPReLUPlugin(float negSlope);
> TRT_DEPRECATED_API INvPlugin* createPReLUPlugin(const void* data, size_t length);
> ```
> 
> In our environment, we are able to build with TensorRT7, but I don't know why...
> Can you run the following command and let me know the results?
> `dpkg -l | grep nvinfer`
> Also, are the results of setup_ubuntu18.04.sh successful without fail?
> If cuda or tensorRT is already installed, I recommend to remove it before setup_ubuntu18.04.sh.
> 
> #### My Environment
> ```
> yukihiro@yukihiro:~/workspace/AutowareArchitectureProposal$ dpkg -l | grep nvinfer
> hi  libnvinfer-dev                               7.0.0-1+cuda10.2                                 amd64        TensorRT development libraries and headers
> hi  libnvinfer-plugin-dev                        7.0.0-1+cuda10.2                                 amd64        TensorRT plugin libraries
> hi  libnvinfer-plugin7                           7.0.0-1+cuda10.2                                 amd64        TensorRT plugin libraries
> hi  libnvinfer7                                  7.0.0-1+cuda10.2                                 amd64        TensorRT runtime libraries
> ```

```
nv@nv-desktop:/usr/local$ dpkg -l | grep nvinfer
ii  libnvinfer-bin                                7.1.0-1+cuda10.2                                 arm64        TensorRT binaries
ii  libnvinfer-dev                                7.1.0-1+cuda10.2                                 arm64        TensorRT development libraries and headers
ii  libnvinfer-doc                                7.1.0-1+cuda10.2                                 all          TensorRT documentation
ii  libnvinfer-plugin-dev                         7.1.0-1+cuda10.2                                 arm64        TensorRT plugin libraries
ii  libnvinfer-plugin7                            7.1.0-1+cuda10.2                                 arm64        TensorRT plugin libraries
ii  libnvinfer-samples                            7.1.0-1+cuda10.2                                 all          TensorRT samples
ii  libnvinfer7                                   7.1.0-1+cuda10.2                                 arm64        TensorRT runtime libraries
ii  python-libnvinfer                             7.1.0-1+cuda10.2                                 arm64        Python bindings for TensorRT
ii  python-libnvinfer-dev                         7.1.0-1+cuda10.2                                 arm64        Python development package for TensorRT
ii  python3-libnvinfer                            7.1.0-1+cuda10.2                                 arm64        Python 3 bindings for TensorRT
ii  python3-libnvinfer-dev                        7.1.0-1+cuda10.2                                 arm64        Python 3 development package for TensorRT
```

sdu2011, 2020-06-23:
> @sdu2011 just installed 7th TensorRT to check if TLR part works as intended:
>
> ```
> dpkg -l | grep TensorRT
> ii  graphsurgeon-tf [...]

```
nv@nv-desktop:/usr/local$ dpkg -l | grep TensorRT
ii  graphsurgeon-tf                               7.1.0-1+cuda10.2                                 arm64        GraphSurgeon for TensorRT package
ii  libnvinfer-bin                                7.1.0-1+cuda10.2                                 arm64        TensorRT binaries
ii  libnvinfer-dev                                7.1.0-1+cuda10.2                                 arm64        TensorRT development libraries and headers
ii  libnvinfer-doc                                7.1.0-1+cuda10.2                                 all          TensorRT documentation
ii  libnvinfer-plugin-dev                         7.1.0-1+cuda10.2                                 arm64        TensorRT plugin libraries
ii  libnvinfer-plugin7                            7.1.0-1+cuda10.2                                 arm64        TensorRT plugin libraries
ii  libnvinfer-samples                            7.1.0-1+cuda10.2                                 all          TensorRT samples
ii  libnvinfer7                                   7.1.0-1+cuda10.2                                 arm64        TensorRT runtime libraries
ii  libnvonnxparsers-dev                          7.1.0-1+cuda10.2                                 arm64        TensorRT ONNX libraries
ii  libnvonnxparsers7                             7.1.0-1+cuda10.2                                 arm64        TensorRT ONNX libraries
ii  libnvparsers-dev                              7.1.0-1+cuda10.2                                 arm64        TensorRT parsers libraries
ii  libnvparsers7                                 7.1.0-1+cuda10.2                                 arm64        TensorRT parsers libraries
ii  nvidia-container-csv-tensorrt                 7.1.0.16-1+cuda10.2                              arm64        Jetpack TensorRT CSV file
ii  python-libnvinfer                             7.1.0-1+cuda10.2                                 arm64        Python bindings for TensorRT
ii  python-libnvinfer-dev                         7.1.0-1+cuda10.2                                 arm64        Python development package for TensorRT
ii  python3-libnvinfer                            7.1.0-1+cuda10.2                                 arm64        Python 3 bindings for TensorRT
ii  python3-libnvinfer-dev                        7.1.0-1+cuda10.2                                 arm64        Python 3 development package for TensorRT
ii  tensorrt                                      7.1.0.16-1+cuda10.2                              arm64        Meta package of TensorRT
ii  uff-converter-tf                              7.1.0-1+cuda10.2                                 arm64        UFF converter for TensorRT package

```

sdu2011, 2020-06-23:
i don't install tensorrt7 on my x86 pc, but i have a docker container with tensorrt6 installed, and i checked header file /usr/include/x86_64-linux-gnu/NvInferPlugin.h,  `createPReLUPlugin` exists in this header file.  

on xavier with tensorrt 7.1 installed, `createPReLUPlugin` does not exists in  /usr/include/x86_64-linux-gnu/NvInferPlugin.h anymore, and functions like 'createXXXPlugin'  are not defined in namespace nvinfer1:plugin anymore.

for now,tensorrt_yolo3 and lidar_apollo_instance_segmentation are all what i need,try to install tensorrt5 on xavier may solve the question(install cuda/tensorrt on xavier is not that easy as on x86 pc...). but i am not sure what other packages i may use in futrue. so it would be great help if tier4 can tell a specific tensorrt version compatible with all autoware packages

wep21, 2020-06-24:
@sdu2011 I think `createPReLUPlugin` was maybe changed to `createLReLUPlugin` in the latest tensorrt update(7.0.0 -> 7.1.3).
However,  `createLReLUPlugin` is also deprecated so I remove this API in #42.

closed 2020-06-24 by wep21