tier4/AutowareArchitectureProposal.proj #41: Training data for Traffic Lights' Fine Detection

sergiev, 2020-06-23:
Thank you for your work!
Could you please describe the data used for training yolov3-tlr.onnx from Fine Detection package?

yukkysaito, 2020-06-25:
Can you comment on this? @taichiH

taichiH, 2020-06-25:
@sergiev 
Thank you for the comment!

> Could you please describe the data used for training yolov3-tlr.onnx from Fine Detection package?

The fine_detection model was trained on the dataset we originally collected in Tokyo. 
If you are thinking about using a trained model, be careful. The input of fine-detection is a larger region of interest created by map-based detection, so it may not work if you input a normal size camera image.

> were there any experience about training YOLO on LISA and using it for both fine detection and light classification?

Yes, we were also experimenting with the LISA dataset! The traffic_light_classifier model on this repository was trained by LISA dataset.  Again, be careful, LISA dataset has labels including arrows, but our classification model was only learning with 3 color labels, so the downloaded label file contains only ["go","stop","warning"] labels.

sergiev, 2020-06-25:
@taichiH thank you for your answer.

> Yes, we were also experimenting with the LISA dataset! The traffic_light_classifier model on this repository was trained by LISA dataset.

I'm familiar with this. I was [fixing the link](https://github.com/tier4/AutowareArchitectureProposal/pull/39) to the dataset in package description. The question is, have you tried to solve both tasks - detection and classification - by using single neural network (pretrained on LISA or anything else) on the whole map-based ROI and if it's so, could you please share your insights on it?

taichiH, 2020-06-26:
sorry, I’ve never tried output bbox and class by YOLO. 
However, I think it is possible because YOLO usually estimates bbox and class at the same time.

In many case, only LampState is different in each region and some time we need to handle irregular type lamp state, so the perception architecture was designed to easy swapping of classifier and is not a detector performance issue.

closed 2020-06-26 by sergiev