tier4/AutowareArchitectureProposal.proj #45: Traffic Light Fine Detection: any special requirements?

sergiev, 2020-06-26:
Hi, I'm using cuda10.2 with tensorRT 7.1.3 to run the traffic_light_fine_detection module. But no matter what, I'm keeping getting the error on this line of code:
```
  CUDA_CHECK(cudaMemcpyAsync(
    mTrtCudaBuffer[inputIndex], inputData, mTrtBindBufferSize[inputIndex], cudaMemcpyHostToDevice,
    mTrtCudaStream));
```
it's from src/traffic_light_fine_detector.cpp, lines 241-243.
Here's how the error looks like:
[image]
* In the first debug line we display the size of input trafficlightroi array.
* In the second line there are coordinates (in order xMin, yMin, xMax, yMax) of rough ROI obtained from map data. 
* Then there are some checks that we're passing exact stages in code. inputData lines shows that we've passed the prepareImage function correctly
* inference stages are about going through particular lines in net_ptr_->doInference.
Here's whole doInference function with its' uncomplicated debug output:
[image]
 
I'll be happy to solve this issue on my own, but my knowledge of CUDA so far limited to capability of swap the cudaMemcpyAsync to cudaMemcpy with respect to documentation. As you may guess, it didn't help :(

taichiH, 2020-06-26:
I’m sorry, It’s our mistake. we’ve attached a wrong download link. we will fix this as soon as possible.

taichiH, 2020-06-28:
@sergiev 
we've fixed the link, please check!
Since this is an architecture proposal version repository (less test), I think the bugs are still somewhere, so please let us know if you find the bugs.

sergiev, 2020-06-29:
thank you so much, @taichiH

closed 2020-06-29 by sergiev

persuelx, 2021-01-08:
you have solved the problem?@sergiev