tier4/AutowareArchitectureProposal.proj #48: just question about Architecture Proposal

Jihun-Kim-kmu, 2020-06-30:
Last time you said I should run it with calibration bag data. so I installed autoware.ai 1.12.0  and tried calibration in autoware
but i  I couldn't find the button in the other video example.(calibration tool kit button).  I just found only calibration pulibsher so I want to konw how to calibration in autoware first. please help me.
and i want to konw that is it possible in autowareArchitectureproposal(calibration)
[image]

taichiH, 2020-07-01:
Hi, thanks for the comment!
You don't need to use autoware.ai to test camera_lidar_fusion_based_detection.

First of all, the [rosbag](https://drive.google.com/file/d/1BFcNjIBUVKwupPByATYczv2X4qZtdAeD/view) doesn't contain images. only velodyne packet( `velodyne_msgs/VelodyneScan`) contained.

I want to help you, so could you tell me what you are trying to do and the sensor environment.

If you use real velodyne and camera, first step is might be
1. launch [euclidean_cluster.launch](https://github.com/tier4/AutowareArchitectureProposal/blob/master/src/perception/object_recognition/detection/euclidean_cluster/launch/euclidean_cluster.launch])  with correct topic name and tf frame name.

Next step is

2. launch camera driver (e.g usb_cam_node)
3. calibration extrinsic and intrinsic camera parameter and velodyne and camera calibration (e.g. static_transform_publisher is easy to use)
4. try camera_lidar_fusion_based_detection

please feel free to contact us!

Jihun-Kim-kmu, 2020-07-01:
>    1. launch camera driver (e.g usb_cam_node)
>    2. calibration extrinsic and intrinsic camera parameter and velodyne
>    3. try camera_lidar_fusion_based_detection

Thank you very much for your kind reply.
My final goal is to get a 3D bounding box by fusing the camera and lidar.
I'm using a vsn360 usb camera and vlp-16 lidar now.
 Which autoware version should I use to create a rosbag file(camera lidar
calibration)?
  Which autoware version should I use to fusion camera and lidar?

taichiH, 2020-07-01:
Thanks for the comment. I seem have to misunderstand.

> Which autoware version should I use to create a rosbag file(camera lidar
calibration)?

I think master version of autoware.ai works calibrate camera and lidar.
https://github.com/Autoware-AI/utilities/tree/master/autoware_camera_lidar_calibrator

> Which autoware version should I use to fusion camera and lidar?

If you have transform between camera link and lidar link in tf tree, you can use this repository.

Jihun-Kim-kmu, 2020-07-01:
Does that mean if i have tf between camera and lidar, I can fusion in
autoware architecture?
Thank you for your help. Can I contact you if there's a problem again?

taichiH, 2020-07-02:
Yes, by default, camera and lidar tf frames must be visible from the base_link frame. 
https://github.com/tier4/AutowareArchitectureProposal/blob/6804847385ec3b683e124ade50d8e03174c08f55/src/perception/object_recognition/detection/euclidean_cluster/launch/euclidean_cluster.launch#L19-L20

so, you need to create tf tree that contains` base_link` to `camera_frame` and `base_link` to `lidar_frame` or change target frame in the launch file to either frame.

Of course! please contact us anytime.

Jihun-Kim-kmu, 2020-07-03:
I installed ubuntu 16.04 and graphic driver, cuda9.0 cudnn7 qt 5.12.2. but
when i tried install master version autoware, occurred error
I want to know that autoware.ai master version have to in ubuntu 18.04??
Is it possible to install and operate autoware.ai master version and
autoware architecture in one computer?

taichiH, 2020-07-03:
@Jihun-Kim-kmu 

please refer to https://github.com/Autoware-AI/autoware.ai/wiki/Source-Build.
master version of autoware.ai does not support ubuntu16.04.

> Is it possible to install and operate autoware.ai master version and
autoware architecture in one computer?

If you have ubuntu18.04 and cuda10.2, yes  you can. 
1. follow [ArchitectureProposalREADME](https://github.com/tier4/AutowareArchitectureProposal) to setup your environment
2. install autoware.ai and build it.

Jihun-Kim-kmu, 2020-07-03:
but I understand that autoware ai and autoware ArchitectureProposal
different cuda version.
autoware have to install cuda 10.
autoware architecture have to install cuda 10.2
Is it possible autoware.ai in cuda 10.2??

taichiH, 2020-07-03:
> Is it possible autoware.ai in cuda 10.2?? 

I think so, in my environment, it is possible to build autoware.ai with cuda 10.2
please try cuda 10.2 and if it is something wrong, please let me know.

Jihun-Kim-kmu, 2020-07-06:
1. install ubuntu 18.04

2. install AutowareArchitectureProposal

(https://github.com/tier4/AutowareArchitectureProposal)

3. play AutowareArchitectureProposal
[image: image.png]

4. install autoware.ai 1.12.0 master version
in this step error occur

$ AUTOWARE_COMPILE_WITH_CUDA=1 colcon build --cmake-args
-DCMAKE_BUILD_TYPE=Release

[image: image.png]
[image: image.png]

Jihun-Kim-kmu, 2020-07-06:
I think error because of cuda version. what do you think?

2020년 7월 6일 (월) 오전 11:38, 김지훈 <wkdrns3847@gmail.com>님이 작성:

taichiH, 2020-07-06:
please show me the console output.
The links you were attached may be invalid.
```
 [image: image.png]
```

Jihun-Kim-kmu, 2020-07-06:
File Attachment (3 format).  All three file attachments are the same.

1. install ubuntu 18.04
2. install AutowareArchitectureProposal
(https://github.com/tier4/AutowareArchitectureProposal)
3. play AutowareArchitectureProposal
roslaunch runtime_manager runtime_manager.launch

4.install autoware.ai 1.12.0 master version
(in this step error occur)
$ AUTOWARE_COMPILE_WITH_CUDA=1 colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release

It is assumed that the concern was caused by the cuda version issue.
So I installed without CUDA support
$ colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release

5. I ran a VLP-16 and a camera as a test.

6. but I can’t find calibrationtool button. Just find calibration pulisher

Q1. What should I do to calibrate it?

taichiH, 2020-07-06:
Sorry, I still can't see your attachment.  Did you create issue comment by e-mail? It may be doing wrong.

> Q1. What should I do to calibrate it?

This is autoware.ai issue, so please create issue on https://github.com/Autoware-AI/autoware.ai/issues

Jihun-Kim-kmu, 2020-07-06:
1. install ubuntu 18.04
2. install AutowareArchitectureProposal
(https://github.com/tier4/AutowareArchitectureProposal)
3. play AutowareArchitectureProposal
roslaunch runtime_manager runtime_manager.launch
[image]
4.install autoware.ai 1.12.0 master version
(in this step error occur)
$ AUTOWARE_COMPILE_WITH_CUDA=1 colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
[image]
[image]
It is assumed that the concern was caused by the cuda version issue.
So I installed without CUDA support
$ colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
[image]
5. I ran a VLP-16 and a camera as a test.
[image]

6. but I can’t find calibrationtool button. Just find calibration pulisher

Q1. What should I do to calibrate it?

Jihun-Kim-kmu, 2020-07-06:
I'm upload again
https://github.com/tier4/AutowareArchitectureProposal/issues/48

taichiH, 2020-07-06:
Thanks! 

You don't need to run AutowareArchitectureProposal.

Please follow the autoware.ai instruction 
https://autoware.readthedocs.io/en/feature-documentation_rtd/DevelopersGuide/PackagesAPI/sensing/autoware_camera_lidar_calibrator.html#camera-lidar-extrinsic-calibration

And if you have more question about autoware.ai, you'd better to open issue on https://github.com/Autoware-AI/autoware.ai/issues .

Jihun-Kim-kmu, 2020-07-07:
Can you check if I understand?
 1. I have to get camera calibration.yaml by using camera intrinsic
calibration in autoware.ai
 2. I have to get lidar_camera calibration.yaml by using camera-lidar
extrinsic calibration in autoware.ai
 3. I get a rosbag file by using calibration.yaml in autoware.ai
 4. Launch euclidean_cluster.launch in AutowareArchitecture

taichiH, 2020-07-07:
step 1,2 and 4 are ok.

after you get intrinsic and extrinsic parameter,
 - launch camera driver with intrinsic paramter file.
 - create tf tree by using extrinsic paramter ([static_transform_publisher](http://wiki.ros.org/tf#static_transform_publisher) is easy to use). in TierIV architecture proposal, we are using urdf robot description and robot_state_publisher, but I think you don't need to use urdf for your purpose.

Jihun-Kim-kmu, 2020-07-18:
can you look autoware.ai issue again?

taichiH, 2020-07-20:
As @JWhitleyWork said on https://github.com/Autoware-AI/autoware.ai/issues/2339#issuecomment-659705208, please request answer on https://answers.ros.org/questions

closed 2020-07-25 by Jihun-Kim-kmu

AthulGangadharan, 2021-08-23:
Hi, how can we visualize the camera & lidar fusion detected objects? I was not able to find the name of the topic to view in Rviz. Thank you!