tier4/AutowareArchitectureProposal.proj #49: Question about apollo seg detect node

LiyouZhou, 2020-07-03:
I am looking at the [lidar segmentation detection node](https://github.com/tier4/AutowareArchitectureProposal/tree/master/src/perception/object_recognition/detection/lidar_apollo_instance_segmentation) and was wondering if you could help me with the following questions:

- How is the training done?
- I see the models are in caffemodel format. Is there possibility of exporting to a different format? such as onnx or tensorflow.

[Caffe-1](https://github.com/BVLC/caffe/releases) has not released new versions in 3 years and [caffe-2](https://caffe2.ai/) has been absorbed into pytourch. The support for this caffemodel format is dwindling in various frameworks.

taichiH, 2020-07-03:
@yukkysaito or @kosuke55 can you comment on this?

yukkysaito, 2020-07-05:
@LiyouZhou thank you for comment.
>How is the training done?

The original Apollo didn't publish the training code, so we are currently working on the training code by trial and error
https://github.com/kosuke55/train_baiducnn

> I see the models are in caffemodel format. Is there possibility of exporting to a different format? such as onnx or tensorflow.

The save format is caffe, but it is converted to tensorRT at runtime.
In the training code we are creating, we will save it in onnx format

closed 2020-09-28 by LalithVipulananthan