tier4/AutowareArchitectureProposal.proj #50: AutowareArchitectureProposal with LG simulator

ajay1606, 2020-07-15:
Hello everyone,

Has anyone tried connecting the AutowareArchitectureProposal version with LG simulator?

yukkysaito, 2020-07-28:
Sorry for the delay.
We've tried to connect internally, but it will take a bit of time with some improvements on the LG simulator. 
e.g. lidar data model, sensor latency, vehicle controller, etc.

ajay1606, 2020-07-29:
@yukkysaito Thanks for your response, We will wait for the updated version of LG simulator.

yuasabe, 2021-06-11:
Hello, are there any updates here in regards to connecting this version of Autoware with LGSVL?

kenji-miyake, 2021-09-15:
Thank you for your contributions/questions.
I'll close this according to [our contribution guidelines](https://github.com/tier4/AutowareArchitectureProposal.proj/blob/main/CONTRIBUTING.md).

closed 2021-09-15 by kenji-miyake