tier4/AutowareArchitectureProposal.proj #53: obstacle_avoidance_planner

lucianzhong, 2020-08-10:
Can someone explain the codes or the design  architecture of obstacle_avoidance_planner in planning/scenario_planning/lane_driving/motion_planning

About the obstacle_avoidance_planner, I only find the reference paper:"A convex optimization approach to smooth trajectories for motion planning with car-like robots"

k0suke-murakami, 2020-08-17:
The main design is: 
1. Make a costmap.
2. Make constraints for reference points based on the costmap.
3. Smoothing reference points by using elastic band, which is from the paper you referred to.

Hope this is helpful for you.

lucianzhong, 2020-08-19:
@k0suke-murakami 
Thanks for the replay. 
The obstacle_avoidance_planner node has optimized position but not the speed the profile. The car can only stop if there is a a moving obstacle. I not should if I am right or not. Please correct me, if I am wrong.

lucianzhong, 2020-08-19:
@k0suke-murakami
Thanks for the replay.
The obstacle_avoidance_planner node has optimized path but not the speed the profile. The car can only stop if there is a a moving obstacle. I not should if I am right or not. Please correct me, if I am wrong.

lucianzhong, 2020-08-19:
@k0suke-murakami
Thanks for the replay.
The obstacle_avoidance_planner node has optimized path but not the speed profile. The car can only stop if there is a moving obstacle. I not should if I am right or not. Please correct me, if I am wrong.

k0suke-murakami, 2020-08-21:
> The obstacle_avoidance_planner node has optimized path but not the speed profile. 

Yes.

The car stop if an "obstacle" is detected on the trajectory. `obstacle_stop_planner`  node is responsible for this function.

lucianzhong, 2020-08-24:
@k0suke-murakami
Thanks a lot for the replay.

lucianzhong, 2020-09-04:
@k0suke-murakami
Hey, Murakami, in your github respositories, there are qp_planner, eb_planner. I am wondering these planners are developed for AutowareArchitectureProposal or not?

k0suke-murakami, 2020-10-20:
> I am wondering these planners are developed for AutowareArchitectureProposal or not?
No, and I would say AutowareArchitectureProposal has a better implementation.

closed 2020-10-26 by LalithVipulananthan