tier4/AutowareArchitectureProposal.proj #54: Map based prediction bugs and slow work

ramizka, 2020-08-12:
Hi. 

There are many problems with map_bases_prediction. 

1. Slow execution time for preparing paths for prediction. 25-30 ms for 15-20 objects just to prepare paths.
2. Strange lanelet searching: if we have only one long lanelet, it's not included in graph. Lanelet size = 0 in LaneletPaths
4. Too much points extracted within paths for interpolating (for example, 140 points)
3. Very slow while interpolating points. For example, we have extracted path with 140 points, interpolating using Spline created 1800 points, and it took 350 ms for only one path per object. 

I had to limit number of points, using distance maximum accumulated distance 200 meters after object and 50 meters before. But it's just a fix to improve speed of interpolating. Can we just use linear interpolating? Or something like this.

k0suke-murakami, 2020-08-17:
- Interpolation takes some time.
  -   Will replace with fast one in the future.
-  Lanelet search fails if you have really long lanelet.
   -   Will modify lanelet util functions in the future.

Thank you for reporting.

kenji-miyake, 2021-09-15:
Thank you for your contributions/questions.
I'll close this according to [our contribution guidelines](https://github.com/tier4/AutowareArchitectureProposal.proj/blob/main/CONTRIBUTING.md).

closed 2021-09-15 by kenji-miyake