tier4/AutowareArchitectureProposal.proj #55: Installation problem

vehxianfish, 2020-08-17:
when I install cuda,the error occurs:
```
TASK [cuda : CUDA (install CUDA 10.2)] *****************************************     
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Failed to update apt cache."}       
	to retry, use: --limit @/home/AutowareArchitectureProposal/ansible/localhost-setup-ubuntu18.04-devpc.retry  

PLAY RECAP *********************************************************************     
localhost                  : ok=3    changed=0    unreachable=0    failed=1       

Complete      
```

wep21, 2020-08-18:
Can you execute `sudo apt update` and try again?
Ansible apt module sometimes fails its task.

LalithVipulananthan, 2020-09-28:
@vehxianfish Could you confirm if you were able to resolve this issue, or if any further assistance is required? :)

0-0-07, 2020-10-12:
> @vehxianfish Could you confirm if you were able to resolve this issue, or if any further assistance is required? 🙂

please help me :) @LalithVipulananthan #75

closed 2020-10-13 by LalithVipulananthan