tier4/AutowareArchitectureProposal.proj #56: Localization with real vehcile data

ajay1606, 2020-08-21:
Hello Everyone,

Thanks for new architecture open source. I would like to test localization with our own vehicle data set to compare the performance with Autoware AI (1.14). But i am getting error in running localization module.
Input topic:

Topic: /points_raw (Single Velodyne 64 lidar)
Topic: /vehicle/twist (Type: geometry_msgs/TwistStamped, frame_id: "base_link")
Topic: /vehicle/imu/data_raw (Type: sensor_msgs/Imu, frame_id: "base_link")
Topic: /sensing/gnss/ (currently disabled and testing with manual RVIZ initialization)

** Launch files modified to receive the input topics 

Step 1: Map loaded

[image]

Step 2: Run rosbag (After running bag file, why /tf position changed ?
[image]

Step 3: Initial Position set (initial position set from rviz not working)
[image]

Error and Warning message;

[image]

Log file info: (localization-util-voxel_grid_filter-30-stdout.log)
[0m[ INFO] [1597976345.144166050]: waitForService: Service [/localization/pose_estimator/ndt_align_srv] has not been advertised, waiting...[0m

Am i missing something !!
Any help to solve the problem much appreciated,

Regards,
Ajay

ajay1606, 2020-09-05:
@yukkysaito  could you please comment on this

yukkysaito, 2020-10-01:
@ajay1606 sorry for late reply. Is it localizationn fail?
Could you please give us a rosbag or video if possible?

ajay1606, 2020-10-08:
@yukkysaito Thanks for your response, Issue has been solved. We are currently using single LiDAR , but the current sensing module configured to 3 Lidar, we have to skip few preprocessing steps. After that its working well. And, we understood that, Lidar distortion removal algorithm currently supported for only 16 and 32 channel lidar, Not for 64 ?

LalithVipulananthan, 2020-10-13:
@ajay1606 So that we can better respond to your query, could you confirm if there is any specific documentation or statement from a developer that says that the distortion correction preprocessor is only supported for 16 and 32 channel LiDAR sensors?

ajay1606, 2020-10-14:
@LalithVipulananthan Thanks for your response, We noticed that the current version of AAP includes few modifications in the Velodyne driver package also. In the file, func.cc (sensing/driver/velodyne/velodyne_pointcloud/src/conversions) it is mentioned that node is debugged only for VLP16 and VLP32C, So is it only for calibrating lidar or including interpolating points ?

LalithVipulananthan, 2020-10-16:
> So is it only for calibrating lidar or including interpolating points ?
It is for interpolation, scan cutting, and pre-filtering. If those functions are required, then only VLP16, VLP-32C, and VLS-128 LiDARs are supported. VLP16 and VLP-32C have been thoroughly tested, and VLS-128 has been tested so far as confirming that the appropriate topic (`velodyne_points_invalid_near`) and points are published. 

Going back to the original issue of localization failing at the original input rate, the processing for interpolation/distortion correction adds extra data to the PointCloud messages and thus the messages are "heavier" than those created by the stock VLS driver.

ajay1606, 2020-10-19:
@LalithVipulananthan Thanks for your response. got it

closed 2020-10-19 by ajay1606