tier4/AutowareArchitectureProposal.proj #57: CMAKE error with CUBLAS_LIBRARIES

ajay1606, 2020-09-04:
Hello ,

I am getting error with fresh installation of Autoware Proposal Architecture, as below please could anyone help me to fix this.

Getting error message after running command;
catkin build --cmake-args -DCMAKE_BUILD_TYPE=Release

Error Message:
[image]

[image]

Environment:
Ubuntu 18.04
ROS: Melodic
Cmake: 3.18.1

Could you please suggest me, Am i missing something ?

yukke42, 2020-09-14:
I got the same error, now I can build successfully.

In my case, I changed 
```
find_library(CUBLAS_LIBRARIES cublas HINTS
  ${CUDA_TOOLKIT_ROOT_DIR}/lib)
```
to
```
find_library(CUBLAS_LIBRARIES cublas HINTS
  ${CUDA_TOOLKIT_ROOT_DIR}/lib64)
```
for the four files.
- src/perception/object_recognition/detection/lidar_apollo_instance_segmentation/CMakeLists.txt
- src/perception/object_recognition/detection/tensorrt_yolo3/CMakeLists.txt
- src/perception/traffic_light_recognition/traffic_light_classifier/CMakeLists.txt
- src/perception/traffic_light_recognition/traffic_light_fine_detector/CMakeLists.txt

In my environment, there is no directory `/usr/local/cuda/lib`.
```
$ ls /usr/local/cuda/lib
ls: cannot access '/usr/local/cuda/lib': No such file or directory
$ echo $LD_LIBRARY_PATH
/opt/ros/melodic/lib:/usr/local/cuda/lib64:
```

ajay1606, 2020-09-14:
@yukke42 Thank you so much

LalithVipulananthan, 2020-09-28:
@ajay1606 Thank you for reporting this issue and thank you @yukke42 for providing an initial workaround. The relevant CMakeLists files have been updated and so this issue should no longer occur.

closed 2020-09-28 by LalithVipulananthan