tier4/AutowareArchitectureProposal.proj #58: Problem with installation

ga74kud, 2020-09-09:
./setup_ubuntu18.04.sh
>  Do you run setup? This will take a while. [y/N] y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ansible is already the newest version (2.5.1+dfsg-1ubuntu0.1).
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 libopenni-sensor-pointclouds0 : Depends: openni-utils but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
 [WARNING]: Unable to parse
/home/hartmannadm/Autoware/AutowareArchitectureProposal/inventories/local-
dev.ini as an inventory source

Do you install GPU modules(cuda: 10.2, cudnn: 7.6.5, TensorRT: 7.0.0)? (y/n) [y]: y

PLAY [localhost] ***************************************************************

TASK [Gathering Facts] *********************************************************
ok: [localhost]

TASK [pause] *******************************************************************
skipping: [localhost]

TASK [cuda : CUDA (add nvidia key)] ********************************************
ok: [localhost]

TASK [cuda : CUDA (install CUDA 10.2 repository)] ******************************
ok: [localhost]

TASK [cuda : CUDA (install CUDA 10.2)] *****************************************
ok: [localhost]

TASK [cuda : CUDA (add path >> bashrc)] ****************************************
ok: [localhost]

TASK [cuda : CUDA (add library >> bashrc)] *************************************
ok: [localhost]

TASK [tensorrt : TensorRT (install machine learning repository)] ***************
ok: [localhost]

TASK [tensorrt : TensorRT (install libraries for TensorRT 7)] ******************
ok: [localhost]

TASK [tensorrt : TensorRT (prevent cuda related packages from upgrading)] ******
changed: [localhost] => (item=cuda-repo-ubuntu1804)
changed: [localhost] => (item=nvidia-machine-learning-repo-ubuntu1804)
changed: [localhost] => (item=cuda-10-2)
changed: [localhost] => (item=libcudnn7)
changed: [localhost] => (item=libcudnn7-dev)
changed: [localhost] => (item=libnvinfer7)
changed: [localhost] => (item=libnvinfer-dev)
changed: [localhost] => (item=libnvinfer-plugin7)
changed: [localhost] => (item=libnvinfer-plugin-dev)
changed: [localhost] => (item=libnvonnxparsers7)
changed: [localhost] => (item=libnvonnxparsers-dev)
changed: [localhost] => (item=libnvparsers7)
changed: [localhost] => (item=libnvparsers-dev)

TASK [ros : ROS (set up your keys)] ********************************************
ok: [localhost]

TASK [ros : ROS (add source list)] *********************************************
ok: [localhost]

TASK [ros : ROS (install packages)] ********************************************
fatal: [localhost]: FAILED! => {"cache_update_time": 1599648403, "cache_updated": true, "changed": false, "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"     install 'ros-melodic-desktop-full'' failed: E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).\n", "rc": 100, "stderr": "E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).\n", "stderr_lines": ["E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution)."], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nYou might want to run 'apt --fix-broken install' to correct these.\nThe following packages have unmet dependencies:\n libopenni-sensor-pointclouds0 : Depends: openni-utils but it is not going to be installed\n ros-melodic-desktop-full : Depends: ros-melodic-desktop but it is not going to be installed\n                            Depends: ros-melodic-perception but it is not going to be installed\n                            Depends: ros-melodic-simulators but it is not going to be installed\n                            Depends: ros-melodic-urdf-sim-tutorial but it is not going to be installed\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "You might want to run 'apt --fix-broken install' to correct these.", "The following packages have unmet dependencies:", " libopenni-sensor-pointclouds0 : Depends: openni-utils but it is not going to be installed", " ros-melodic-desktop-full : Depends: ros-melodic-desktop but it is not going to be installed", "                            Depends: ros-melodic-perception but it is not going to be installed", "                            Depends: ros-melodic-simulators but it is not going to be installed", "                            Depends: ros-melodic-urdf-sim-tutorial but it is not going to be installed"]}
	to retry, use: --limit @/home/hartmannadm/Autoware/AutowareArchitectureProposal/ansible/localhost-setup-ubuntu18.04-devpc.retry

PLAY RECAP *********************************************************************
localhost                  : ok=11   changed=1    unreachable=0    failed=1   

Complete

ga74kud, 2020-09-09:
sudo apt --fix-broken install -f
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following additional packages will be installed:
  openni-utils
The following NEW packages will be installed:
  openni-utils
0 upgraded, 1 newly installed, 0 to remove and 306 not upgraded.
1 not fully installed or removed.
Need to get 0 B/115 kB of archives.
After this operation, 474 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
(Reading database ... 420926 files and directories currently installed.)
Preparing to unpack .../openni-utils_1.5.4.0-14build1_amd64.deb ...
Unpacking openni-utils (1.5.4.0-14build1) ...
dpkg: error processing archive /var/cache/apt/archives/openni-utils_1.5.4.0-14build1_amd64.deb (--unpack):
 trying to overwrite '/usr/bin/NiViewer', which is also in package nuitrack 0.30.0
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/openni-utils_1.5.4.0-14build1_amd64.deb
libdvd-pkg: Checking orig.tar integrity...
/usr/src/libdvd-pkg/libdvdcss_1.4.2.orig.tar.bz2: OK
libdvd-pkg: `apt-get check` failed, you may have broken packages. Aborting...
E: Sub-process /usr/bin/dpkg returned an error code (1)

ga74kud, 2020-09-09:
similar problem
 https://community.nuitrack.com/t/solved-how-to-resolve-conflict-between-nuitrack-and-ros-packages-problem-with-openni-utils/2115

ga74kud, 2020-09-09:
I tried  apt-get -o Dpkg::Options::="--force-overwrite" -f install from https://answers.ros.org/question/193581/libopenni-sensor-primesense0-conflict/ solved unmet dependencies

ga74kud, 2020-09-09:
rosdep install http://wiki.ros.org/rosdep

ga74kud, 2020-09-09:
sudo apt-get install python-rosdep
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-rosdep is already the newest version (0.19.0-1).
0 upgraded, 0 newly installed, 0 to remove and 302 not upgraded.
(base) hartmannadm@insc02303:~/Autoware/AutowareArchitectureProposal$ sudo pip install -U rosdep
The directory '/home/hartmannadm/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/hartmannadm/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already up-to-date: rosdep in /usr/lib/python2.7/dist-packages
Collecting PyYAML>=3.1 (from rosdep)
  Downloading https://files.pythonhosted.org/packages/64/c2/b80047c7ac2478f9501676c988a5411ed5572f35d1beff9cae07d321512c/PyYAML-5.3.1.tar.gz (269kB)
    100% |████████████████████████████████| 276kB 2.3MB/s 
Collecting catkin_pkg>=0.4.0 (from rosdep)
  Downloading https://files.pythonhosted.org/packages/96/70/0f6d4c2a6b251d01ef626c4b33a4b5347a1af9be8c75ce1bfa98d51705ee/catkin_pkg-0.4.22.tar.gz (62kB)
    100% |████████████████████████████████| 71kB 5.2MB/s 
Collecting rosdistro>=0.7.5 (from rosdep)
  Downloading https://files.pythonhosted.org/packages/88/28/160b8af1f59f8f991a3509b39141142b39b1d3e4aabf31cb7ae2d6597c72/rosdistro-0.8.2.tar.gz (47kB)
    100% |████████████████████████████████| 51kB 5.4MB/s 
Requirement already up-to-date: rospkg>=1.1.10 in /usr/lib/python2.7/dist-packages (from rosdep)
Collecting docutils (from catkin_pkg>=0.4.0->rosdep)
  Downloading https://files.pythonhosted.org/packages/81/44/8a15e45ffa96e6cf82956dd8d7af9e666357e16b0d93b253903475ee947f/docutils-0.16-py2.py3-none-any.whl (548kB)
    100% |████████████████████████████████| 552kB 1.9MB/s 
Collecting python-dateutil (from catkin_pkg>=0.4.0->rosdep)
  Downloading https://files.pythonhosted.org/packages/d4/70/d60450c3dd48ef87586924207ae8907090de0b306af2bce5d134d78615cb/python_dateutil-2.8.1-py2.py3-none-any.whl (227kB)
    100% |████████████████████████████████| 235kB 3.0MB/s 
Collecting pyparsing (from catkin_pkg>=0.4.0->rosdep)
  Downloading https://files.pythonhosted.org/packages/8a/bb/488841f56197b13700afd5658fc279a2025a39e22449b7cf29864669b15d/pyparsing-2.4.7-py2.py3-none-any.whl (67kB)
    100% |████████████████████████████████| 71kB 5.7MB/s 
Collecting setuptools (from rosdistro>=0.7.5->rosdep)
  Downloading https://files.pythonhosted.org/packages/e1/b7/182161210a13158cd3ccc41ee19aadef54496b74f2817cc147006ec932b4/setuptools-44.1.1-py2.py3-none-any.whl (583kB)
    100% |████████████████████████████████| 583kB 2.0MB/s 
Collecting six>=1.5 (from python-dateutil->catkin_pkg>=0.4.0->rosdep)
  Downloading https://files.pythonhosted.org/packages/ee/ff/48bde5c0f013094d729fe4b0316ba2a24774b3ff1c52d924a8a4cb04078a/six-1.15.0-py2.py3-none-any.whl
Installing collected packages: PyYAML, docutils, six, python-dateutil, pyparsing, catkin-pkg, setuptools, rosdistro
  Found existing installation: PyYAML 3.12
    Not uninstalling pyyaml at /usr/lib/python2.7/dist-packages, outside environment /usr
  Running setup.py install for PyYAML ... done
  Found existing installation: docutils 0.14
    Not uninstalling docutils at /usr/lib/python2.7/dist-packages, outside environment /usr
  Found existing installation: six 1.12.0
    Uninstalling six-1.12.0:
      Successfully uninstalled six-1.12.0
  Found existing installation: python-dateutil 2.8.0
    Uninstalling python-dateutil-2.8.0:
      Successfully uninstalled python-dateutil-2.8.0
  Found existing installation: pyparsing 2.4.0
    Uninstalling pyparsing-2.4.0:
      Successfully uninstalled pyparsing-2.4.0
  Found existing installation: catkin-pkg 0.4.12
    Not uninstalling catkin-pkg at /usr/lib/python2.7/dist-packages, outside environment /usr
  Running setup.py install for catkin-pkg ... done
  Found existing installation: setuptools 44.0.0
    Uninstalling setuptools-44.0.0:
      Successfully uninstalled setuptools-44.0.0
  Found existing installation: rosdistro 0.7.4
    Not uninstalling rosdistro at /usr/lib/python2.7/dist-packages, outside environment /usr
  Running setup.py install for rosdistro ... done
Successfully installed PyYAML-5.3.1 catkin-pkg-0.4.22 docutils-0.16 pyparsing-2.4.7 python-dateutil-2.8.1 rosdistro-0.8.2 setuptools-44.1.1 six-1.15.0
(base) hartmannadm@insc02303:~/Autoware/AutowareArchitectureProposal$ ./setup_ubuntu18.04.sh
>  Do you run setup? This will take a while. [y/N] y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ansible is already the newest version (2.5.1+dfsg-1ubuntu0.1).
0 upgraded, 0 newly installed, 0 to remove and 302 not upgraded.
 [WARNING]: Unable to parse /home/hartmannadm/Autoware/AutowareArchitectureProposal/inventories/local-dev.ini as an inventory source

Do you install GPU modules(cuda: 10.2, cudnn: 7.6.5, TensorRT: 7.0.0)? (y/n) [y]: y

[... 86 lines cut ...]
TASK [osqp : osqp (cmake)] *********************************************************************************************************************************************************************************
changed: [localhost]

[... cut ...]
TASK [autoware : Autoware (install egm2008-1)] *************************************************************************************************************************************************************
changed: [localhost]

PLAY RECAP *************************************************************************************************************************************************************************************************
localhost                  : ok=29   changed=14   unreachable=0    failed=0   

Complete

ga74kud, 2020-09-09:
catkin build --cmake-args -DCMAKE_BUILD_TYPE=Release

--------------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/melodic
Workspace:                   /home/hartmannadm/Autoware/AutowareArchitectureProposal
--------------------------------------------------------------------------------
Build Space:        [exists] /home/hartmannadm/Autoware/AutowareArchitectureProposal/build
Devel Space:        [exists] /home/hartmannadm/Autoware/AutowareArchitectureProposal/devel
Install Space:      [unused] /home/hartmannadm/Autoware/AutowareArchitectureProposal/install
Log Space:         [missing] /home/hartmannadm/Autoware/AutowareArchitectureProposal/logs
Source Space:       [exists] /home/hartmannadm/Autoware/AutowareArchitectureProposal/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
--------------------------------------------------------------------------------
[build] Found '104' packages in 0.0 seconds.                                   
[build] Updating package table.                                                
Starting  >>> catkin_tools_prebuild                                            
_______________________________________________________________________________
Errors     << catkin_tools_prebuild:cmake /home/hartmannadm/Autoware/AutowareArchitectureProposal/logs/catkin_tools_prebuild/build.cmake.000.log
/home/hartmannadm/Autoware/AutowareArchitectureProposal/devel/.private/catkin_tools_prebuild/env.sh: 31: /tmp/setup.sh.lueCFaD89R: Syntax error: Unterminated quoted string
Traceback (most recent call last):
  File "/home/hartmannadm/Autoware/AutowareArchitectureProposal/build/catkin_tools_prebuild/catkin_generated/generate_cached_setup.py", line 22, in <module>
    code = generate_environment_script('/home/hartmannadm/Autoware/AutowareArchitectureProposal/devel/.private/catkin_tools_prebuild/env.sh')
  File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin/environment_cache.py", line 62, in generate_environment_script
    output = subprocess.check_output([env_script, sys.executable, '-c', python_code])
  File "/usr/lib/python2.7/subprocess.py", line 223, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['/home/hartmannadm/Autoware/AutowareArchitectureProposal/devel/.private/catkin_tools_prebuild/env.sh', '/usr/bin/python2', '-c', 'import os; print(dict(os.environ))']' returned non-zero exit status 2
CMake Error at /opt/ros/melodic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/usr/bin/python2
  "/home/hartmannadm/Autoware/AutowareArchitectureProposal/build/catkin_tools_prebuild/catkin_generated/generate_cached_setup.py")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/all.cmake:208 (safe_execute_process)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:4 (find_package)

cd /home/hartmannadm/Autoware/AutowareArchitectureProposal/build/catkin_tools_prebuild; catkin build --get-env catkin_tools_prebuild | catkin env -si  /usr/bin/cmake /home/hartmannadm/Autoware/AutowareArchitectureProposal/build/catkin_tools_prebuild --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/hartmannadm/Autoware/AutowareArchitectureProposal/devel/.private/catkin_tools_prebuild -DCMAKE_INSTALL_PREFIX=/home/hartmannadm/Autoware/AutowareArchitectureProposal/install -DCMAKE_BUILD_TYPE=Release; cd -
...............................................................................
Failed     << catkin_tools_prebuild:cmake                       [ Exited with code 1 ]
Failed    <<< catkin_tools_prebuild                             [ 7.2 seconds ]
Abandoned <<< additional_vehicle_info_generator                 [ Unrelated job failed ]
Abandoned <<< astar_search                                      [ Unrelated job failed ]
Abandoned <<< autoware_control_msgs                             [ Unrelated job failed ]
Abandoned <<< autoware_lanelet2_msgs                            [ Unrelated job failed ]
Abandoned <<< autoware_localization_srvs                        [ Unrelated job failed ]
Abandoned <<< autoware_perception_msgs                          [ Unrelated job failed ]
Abandoned <<< autoware_planning_msgs                            [ Unrelated job failed ]
Abandoned <<< autoware_planning_rviz_plugin                     [ Unrelated job failed ]
Abandoned <<< autoware_rosbag_recorder                          [ Unrelated job failed ]
Abandoned <<< autoware_system_msgs                              [ Unrelated job failed ]
Abandoned <<< autoware_state_monitor                            [ Unrelated job failed ]
[... cut ...]
Abandoned <<< velodyne_pointcloud                               [ Unrelated job failed ]
[build] Summary: 0 of 102 packages succeeded.                                  
[build]   Ignored:   3 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 101 packages were abandoned.                              
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 7.8 seconds total.

ga74kud, 2020-09-09:
gedit /home/hartmannadm/Autoware/AutowareArchitectureProposal/logs/catkin_tools_prebuild/build.cmake.004.log

Not searching for unused variables given on the command line.
[36m--[0m Using CATKIN_DEVEL_PREFIX: /home/hartmannadm/Autoware/AutowareArchitectureProposal/devel/.private/catkin_tools_prebuild
[36m--[0m Using CMAKE_PREFIX_PATH: /opt/ros/melodic
[36m--[0m This workspace overlays: /opt/ros/melodic
[36m--[0m Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2")
[36m--[0m Using PYTHON_EXECUTABLE: /usr/bin/python2
[36m--[0m Using Debian Python package layout
[36m--[0m Using empy: /usr/bin/empy
[36m--[0m Using CATKIN_ENABLE_TESTING: ON
[36m--[0m Call enable_testing()
[36m--[0m Using CATKIN_TEST_RESULTS_DIR: /home/hartmannadm/Autoware/AutowareArchitectureProposal/build/catkin_tools_prebuild/test_results
[36m--[0m Found gtest sources under '/usr/src/googletest': gtests will be built
[36m--[0m Found gmock sources under '/usr/src/googletest': gmock will be built
[36m--[0m Found PythonInterp: /usr/bin/python2 (found version "2.7.17")
[36m--[0m Using Python nosetests: /usr/bin/nosetests-2.7
[36m--[0m catkin 0.7.28
[36m--[0m BUILD_SHARED_LIBS is on
/home/hartmannadm/Autoware/AutowareArchitectureProposal/devel/.private/catkin_tools_prebuild/env.sh: 31: /tmp/setup.sh.GEcxa1AJwL: Syntax error: Unterminated quoted string
Traceback (most recent call last):
  File "/home/hartmannadm/Autoware/AutowareArchitectureProposal/build/catkin_tools_prebuild/catkin_generated/generate_cached_setup.py", line 22, in <module>
    code = generate_environment_script('/home/hartmannadm/Autoware/AutowareArchitectureProposal/devel/.private/catkin_tools_prebuild/env.sh')
  File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin/environment_cache.py", line 62, in generate_environment_script
    output = subprocess.check_output([env_script, sys.executable, '-c', python_code])
  File "/usr/lib/python2.7/subprocess.py", line 223, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['/home/hartmannadm/Autoware/AutowareArchitectureProposal/devel/.private/catkin_tools_prebuild/env.sh', '/usr/bin/python2', '-c', 'import os; print(dict(os.environ))']' returned non-zero exit status 2
[31m[1mCMake Error[0m at /opt/ros/melodic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/usr/bin/python2
  "/home/hartmannadm/Autoware/AutowareArchitectureProposal/build/catkin_tools_prebuild/catkin_generated/generate_cached_setup.py")
  returned error code 1
[36mCall Stack (most recent call first):[0m
  /opt/ros/melodic/share/catkin/cmake/all.cmake:208 (safe_execute_process)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:4 (find_package)

[36m--[0m Configuring incomplete, errors occurred!
See also "/home/hartmannadm/Autoware/AutowareArchitectureProposal/build/catkin_tools_prebuild/CMakeFiles/CMakeOutput.log".
See also "/home/hartmannadm/Autoware/AutowareArchitectureProposal/build/catkin_tools_prebuild/CMakeFiles/CMakeError.log".

ga74kud, 2020-09-09:
catkin and anaconda problem https://stackoverflow.com/questions/51134709/catkin-build-fails-to-compile

and a problem with ENV variable https://answers.ros.org/question/347743/unexpected-eof/ solved the issue

ga74kud, 2020-09-09:
Finished  <<< ndt                                               [ 27.0 seconds ]
[build] Summary: 74 of 101 packages succeeded.                                 
[build]   Ignored:   3 packages were skipped or are blacklisted.               
[build]   Warnings:  2 packages succeeded with warnings.                       
[build]   Abandoned: 26 packages were abandoned.                               
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 38.5 seconds total.                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.

ga74kud, 2020-09-09:
https://stackoverflow.com/questions/44633043/cmake-libcurl-was-built-with-ssl-disabled-https-not-supported/44633762

ga74kud, 2020-09-09:
https://stackoverflow.com/questions/53205687/ansible-unable-to-parse-etc-ansible-hosts-as-an-inventory-source

ga74kud, 2020-09-09:
nvidia-smi
Wed Sep  9 15:34:46 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.33.01    Driver Version: 440.33.01    CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce RTX 208...  On   | 00000000:65:00.0  On |                  N/A |
| 35%   36C    P8    33W / 250W |   1044MiB / 11016MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0       759      G   ...-2019.3.3/pycharm-2019.3.3/jbr/bin/java     8MiB |
|    0      1992      G   /usr/lib/xorg/Xorg                            26MiB |
|    0      2074      G   /usr/bin/gnome-shell                          57MiB |
|    0      3276      G   /usr/lib/firefox/firefox                       6MiB |
|    0      3374      G   /usr/lib/xorg/Xorg                           461MiB |
|    0      3539      G   /usr/bin/gnome-shell                         371MiB |
|    0      4045      G   ...uest-channel-token=10713016311349530732    76MiB |
|    0      4670      G   /usr/lib/firefox/firefox                       6MiB |
|    0      7589      G   /usr/lib/firefox/firefox                       6MiB |
|    0     13384      G   /usr/lib/firefox/firefox                       6MiB |
|    0     23435      G   /usr/lib/firefox/firefox                       6MiB |
|    0     32199      G   /usr/lib/firefox/firefox                       6MiB |
+-----------------------------------------------------------------------------+

ga74kud, 2020-09-09:
https://github.com/NVIDIA/TensorRT/issues/336

ga74kud, 2020-09-09:
https://github.com/tensorflow/tensorflow/issues/34329

ga74kud, 2020-09-09:
it seems that tensorrt is not installed by the autoware install

 python3
Python 3.6.9 (default, Jul 17 2020, 12:50:27) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tensorrt
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'tensorrt'
>>> exit()
hartmannadm@insc02303:~/Autoware/AutowareArchitectureProposal$ python
Python 2.7.17 (default, Jul 20 2020, 15:37:01) 
[GCC 7.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import tensorrt
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named tensorrt
>>> 
KeyboardInterrupt
>>> 
KeyboardInterrupt
>>> exit()
hartmannadm@insc02303:~/Autoware/AutowareArchitecturePropos

ga74kud, 2020-09-10:
tensorrt installed
hartmannadm@insc02303:~$ python 
Python 2.7.17 (default, Jul 20 2020, 15:37:01) 
[GCC 7.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import tensorrt

ga74kud, 2020-09-10:
hartmannadm@insc02303:~/Autoware/AutowareArchitectureProposal$ ./setup_ubuntu18.04.sh
>  Do you run setup? This will take a while. [y/N] y
[sudo] password for hartmannadm: 
Your password has expired
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ansible is already the newest version (2.5.1+dfsg-1ubuntu0.1).
0 upgraded, 0 newly installed, 0 to remove and 299 not upgraded.
 [WARNING]: Unable to parse
/home/hartmannadm/Autoware/AutowareArchitectureProposal/inventories/local-
dev.ini as an inventory source

Do you install GPU modules(cuda: 10.2, cudnn: 7.6.5, TensorRT: 7.0.0)? (y/n) [y]: y

PLAY [localhost] ***************************************************************

TASK [Gathering Facts] *********************************************************
ok: [localhost]

TASK [pause] *******************************************************************
skipping: [localhost]

TASK [cuda : CUDA (add nvidia key)] ********************************************
ok: [localhost]

TASK [cuda : CUDA (install CUDA 10.2 repository)] ******************************
ok: [localhost]

TASK [cuda : CUDA (install CUDA 10.2)] *****************************************
ok: [localhost]

TASK [cuda : CUDA (add path >> bashrc)] ****************************************
ok: [localhost]

TASK [cuda : CUDA (add library >> bashrc)] *************************************
ok: [localhost]

TASK [tensorrt : TensorRT (install machine learning repository)] ***************
ok: [localhost]

TASK [tensorrt : TensorRT (install libraries for TensorRT 7)] ******************
ok: [localhost]

TASK [tensorrt : TensorRT (prevent cuda related packages from upgrading)] ******
changed: [localhost] => (item=cuda-repo-ubuntu1804)
changed: [localhost] => (item=nvidia-machine-learning-repo-ubuntu1804)
changed: [localhost] => (item=cuda-10-2)
changed: [localhost] => (item=libcudnn7)
changed: [localhost] => (item=libcudnn7-dev)
changed: [localhost] => (item=libnvinfer7)
changed: [localhost] => (item=libnvinfer-dev)
changed: [localhost] => (item=libnvinfer-plugin7)
changed: [localhost] => (item=libnvinfer-plugin-dev)
changed: [localhost] => (item=libnvonnxparsers7)
changed: [localhost] => (item=libnvonnxparsers-dev)
changed: [localhost] => (item=libnvparsers7)
changed: [localhost] => (item=libnvparsers-dev)

TASK [ros : ROS (set up your keys)] ********************************************
ok: [localhost]

TASK [ros : ROS (add source list)] *********************************************
ok: [localhost]

TASK [ros : ROS (install packages)] ********************************************
ok: [localhost]

TASK [ros : ROS (check rosdep file)] *******************************************
ok: [localhost]

TASK [ros : ROS (install rosdep)] **********************************************
ok: [localhost]

TASK [ros : ROS (initilize rosdep)] ********************************************
skipping: [localhost]

TASK [ros : ROS (update rosdep)] ***********************************************
changed: [localhost]

TASK [ros : ROS (add path >> bashrc)] ******************************************
ok: [localhost]

TASK [ros : ROS (install dependencies for building packages)] ******************
ok: [localhost]

TASK [lanelet2 : Lanelet2 (install dependencies for building packages)] ********
ok: [localhost]

TASK [lanelet2 : Lanelet2 (clone Lanelet2 repository catkin workspace)] ********
ok: [localhost]

TASK [lanelet2 : Lanelet2 (clone mrt_cmake_modules catkin workspace)] **********
ok: [localhost]

TASK [osqp : osqp (clone repository)] ******************************************
ok: [localhost]

TASK [osqp : osqp (make build dir)] ********************************************
ok: [localhost] => (item=build)

TASK [osqp : osqp (cmake)] *****************************************************
changed: [localhost]

TASK [osqp : osqp (make all and install)] **************************************
changed: [localhost]

TASK [autoware : Autoware (update rosdep)] *************************************
changed: [localhost]

TASK [autoware : Autoware (install dependencies)] ******************************
changed: [localhost]

TASK [autoware : Autoware (install pip)] ***************************************
ok: [localhost]

TASK [autoware : Autoware (install gdown to download some files for neural network)] ***
changed: [localhost]

TASK [autoware : Autoware (install egm2008-1)] *********************************
changed: [localhost]

PLAY RECAP *********************************************************************
localhost                  : ok=29   changed=8    unreachable=0    failed=0   

Complete

ga74kud, 2020-09-10:
Errors     << tensorrt_yolo3:make /home/hartmannadm/Autoware/AutowareArchitectureProposal/logs/tensorrt_yolo3/build.make.007.log
In file included from /usr/local/cuda/include/channel_descriptor.h:61:0,
                 from /usr/local/cuda/include/cuda_runtime.h:95,
                 from /usr/local/cuda/include/cudnn.h:64,
                 from /home/hartmannadm/Autoware/AutowareArchitectureProposal/src/perception/object_recognition/detection/tensorrt_yolo3/lib/include/YoloLayer.h:29,
                 from /home/hartmannadm/Autoware/AutowareArchitectureProposal/src/perception/object_recognition/detection/tensorrt_yolo3/lib/src/YoloLayer.cpp:25:
/usr/local/cuda/include/cuda_runtime_api.h:2297:25: error: expected ‘)’ before ‘*’ token
 typedef void (CUDART_CB *cudaStreamCallback_t)(cudaStream_t stream, cudaError_t status, void *userData);
                         ^
/usr/local/cuda/include/cuda_runtime_api.h:2365:9: error: ‘cudaStreamCallback_t’ has not been declared
         cudaStreamCallback_t callback, void *userData, unsigned int flags);
         ^~~~~~~~~~~~~~~~~~~~
/usr/local/cuda/include/cuda_runtime_api.h:2532:88: error: use of enum ‘cudaStreamCaptureMode’ without previous declaration
 extern __host__ cudaError_t CUDARTAPI cudaStreamBeginCapture(cudaStream_t stream, enum cudaStreamCaptureMode mode);

ga74kud, 2020-09-10:
[build] Summary: 75 of 101 packages succeeded.                                 
[build]   Ignored:   3 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 25 packages were abandoned.                               
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 16.3 seconds total.

ga74kud, 2020-09-10:
seems that yolo3 to be installed

closed 2020-09-10 by ga74kud

ga74kud, 2020-09-10:
added a CATKIN_IGNORE to the folder

0-0-07, 2020-10-12:
@ga74kud Hello, if you have time, can you help me? Many thanks #75