tier4/AutowareArchitectureProposal.proj #65: Porting AutowareArchitectureProposal to ROS2

mitsudome-r, 2020-10-06:
1. Short intro
2. R/K presents the https://github.com/tier4/AutowareArchitectureProposal
3. R/K list the packages which we should port in until Oct 12
4. Communication
    1. where to ask questions about https://github.com/tier4/AutowareArchitectureProposal
        1. respective repositories
    2. where to report bugs and features
        1. respective repositories
    3. where to communicate about everything else (meetings cancellation)
        1. emails via a mailing list
    4. Meeting minutes
        1. https://github.com/tier4/AutowareArchitectureProposal/issues
    5. Work planning (who ports which package by when)
        1. Into the meeting minutes issue
5. Apex.AI ports 1 example package
6. Apex.AI prepares ade for ROS 2 Foxy & github
    1. https://gitlab.com/ApexAI/ade-cli/-/merge_requests/47
7. R presents CI
    1. https://github.com/tier4/autoware.iv.universe/blob/ros2/.github/workflows/build_and_test.yml
    2. https://github.com/tier4/autoware.iv.universe/pull/2
    3. Present how to install ROS 2 Foxy with plain docker
8. Branch name ros2
9. MRs to be done into the github projects listed in https://github.com/tier4/AutowareArchitectureProposal/blob/master/autoware.proj.repos
10. MRs will be reviewed by 5 Prometheus engineers (Simon, Ryohsuke, Jillada, Esteve, Taka)

dejanpan, 2020-10-06:
For vcs tool, the documentation: https://github.com/tier4/AutowareArchitectureProposal/issues/65

dejanpan, 2020-10-06:
## Meeting minutes
1. Run the clang formatter
1. Use the spreadsheet to assign the packages to the developers
1. CI issues to be reported to @mitsudome-r 
1. @esteve and @jpsamper will create the `ade` for ROS 2 Foxy & Github
1.

dejanpan, 2020-10-06:
@simon-t4 a porting guide: https://index.ros.org/doc/ros2/Contributing/Migration-Guide/

nnmm, 2020-10-07:
For reference, the GitHub usernames of the five reviewers are: mitsudome-r, esteve, simon-t4, jilaada, TakaHoribe 
The usernames of Apex.AI contributors are nnmm, fred-apex-ai, cvasfi

closed 2020-10-13 by mitsudome-r