tier4/AutowareArchitectureProposal.proj #68: [ROS2 Porting] Meeting 2020-10-08

mitsudome-r, 2020-10-08:
1. Attendees:
1. Report from Apex about porting to ROS2
1. Share any issues/guidelines during porting
1. Creating fork or creating branch to upstream?
1. Use of external packages

dejanpan, 2020-10-08:
@mitsudome-r 2 things to add:
1. My instructions regarding the documentation (see in slack)
1. Apex.AI will provide the guidance regarding the proper way of using the Timer constructs (Chris is helping)

simon-t4, 2020-10-08:
Apologies - I have another meeting at 4pm. I will try to join later, and from next week I will reschedule the conflicting meeting.
As an update for my assigned package pointcloud_preprocess I have set up the development environment and had a quick look at the package, potential concerns are
dependency on  ros_pcl as being discussed for other nodes
dynamic reconfiguration and .cfg files (all should be normal paramters in ROS2?)
filter nodelets (In ROS2 all should just be components)

nnmm, 2020-10-08:
I asked Steve Macenski about `pcl_ros` [here](https://github.com/ros-perception/perception_pcl/issues/278#issuecomment-705608634), but my impression is that the code isn't ready yet, so vendoring the source code might not work either.

On the plus side, I found out that `pcl_conversions` and plain `pcl` is sufficient for the `pose_initializer` package – so at least there, we don't need `pcl_ros`.

fred-apex-ai, 2020-10-09:
>     * dynamic reconfiguration and .cfg files (all should be normal paramters in ROS2?)
@simon-t4 I have the same concern in porting `mpc_follower` right now. I didn't get to the point yet where I need to change the parameter handling but I also wonder how to test that properly

dejanpan, 2020-10-11:
@simon-t4 re
> * filter nodelets (In ROS2 all should just be components)

Yes, this is the way to port nodelets https://index.ros.org/doc/ros2/Tutorials/Composition/

However we need to asses if this is really ok for production, e.g. Autoware.Auto because of the dynamic loading/unloading. I know that Josh went ahead with it but I want someone to have a 2nd look at this aspect.

dejanpan, 2020-10-12:
@fred-apex-ai @simon-t4 re:

> dynamic reconfiguration and .cfg files (all should be normal paramters in ROS2?)

In short, the equivalent functionality of dynamic reconfiguration(http://wiki.ros.org/dynamic_reconfigure) in ROS2 is ROS2 parameter events and callback (https://index.ros.org/doc/ros2/Tutorials/Parameters/Understanding-ROS2-Parameters/). See https://discourse.ros.org/t/dynamic-reconfigure-porting/6497/3. I quickly looked over the features comparison between `dynamic_configure` and ROS 2 parameters and all of the features are there in ROS2. There is also a `rqt_reconfigure` tool to introspect the parameters with a GUI interface since dashing.

For more info please ping Weihung from Apex.AI.

closed 2020-10-13 by mitsudome-r