tier4/AutowareArchitectureProposal.proj #72: [ROS2 Porting] Meeting 2020-10-12

mitsudome-r, 2020-10-09:
1. Attendees
1. Go over actions items from previous meeting
    - https://github.com/tier4/AutowareArchitectureProposal/issues/68
1. Updates from each engineers:
1. Issues concerns:
    - Use of global parameters
1. Distortion correction for velodyne drivers (left over from previous contract)

dejanpan, 2020-10-11:
@mitsudome-r I am adding these items to the agenda as well:

1. Bringing in external dependencies https://github.com/tier4/Pilot.Auto/pull/14#issuecomment-706725792
2. Dependency resolution in ros packages
    Just shortly how dependency resolution in ros packages works:
    1. you specify the ros dependency in the package.xml, e.g. https://github.com/tier4/Pilot.Auto/blob/9a6ac8109edeeb063d150b0183a02b40589dc50b/localization/util/pose_initializer/package.xml#L18
rosdep command than looks up what is the real name of that dependency for a given OS (e.g. Ubuntu, Mac, ...) via this file https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L3408
See http://wiki.ros.org/ROS/Tutorials/rosdep (it works the same in ROS 1 & ROS 2)
3. Implement distortion correction for the LiDAR stack => this was a left over item from the 2018 contract. 
   1. If we can have a driver that works in ROS2 with the feature, it's fine.
It's either we add the feature to existing driver in Autoware.Auto or port the ones we have in our repo.
4. Questions 2 & 3 from here: https://github.com/tier4/AutowareArchitectureProposal/issues/68#issuecomment-705394505
5. Lets have 3 columns here: in-progress (with the link to the PR), in-review, done
    1. Also move the node dependency picture somewhere else
6. Communication:
    1. We need to make sure that we do not miss pings
    2. Please use @mentions
    3. To read the @mentions use:
        1. https://github.com/notifications?query=reason%3Amention or
        3. create the filter in gmail https://gist.github.com/ldez/bd6e6401ad0855e6c0de6da19a8c50b5#mentions
7. Timer resolution:
    1. https://github.com/tier4/Pilot.Auto/pull/3#issuecomment-706732396

dejanpan, 2020-10-13:
@mitsudome-r this is how you re-construct the `global` parameter use case from ROS 1: https://index.ros.org/doc/ros2/Tutorials/Node-arguments/#setting-parameters-from-yaml-files.

So you use a stand-alone node like this one here: https://github.com/ros2/demos/blob/master/demo_nodes_cpp/src/parameters/parameter_blackboard.cpp. This allows you to introspect and also set the parameters via ROS 2 services.

This approach or approach of having the `global` parameter server also comes with the caveat which Tully sums very well here: https://discourse.ros.org/t/ros2-global-parameter-server-status/10114/10.

closed 2020-10-15 by mitsudome-r