tier4/AutowareArchitectureProposal.proj #75: Installation problem

0-0-07, 2020-10-12:
The result of running  ./setup_ubuntu18.04.sh
~
~
~
ubuntu1804@ubuntu1804-VirtualBox:~$ sudo apt update 
[sudo] ubuntu1804 的密码： 
命中:1 http://mirrors.ustc.edu.cn/ros/ubuntu bionic InRelease
命中:2 https://mirrors.ustc.edu.cn/ubuntu bionic InRelease                     
命中:3 https://mirrors.ustc.edu.cn/ubuntu bionic-updates InRelease             
命中:4 http://linux.teamviewer.com/deb stable InRelease                        
命中:5 https://mirrors.ustc.edu.cn/ubuntu bionic-backports InRelease           
命中:6 https://mirrors.ustc.edu.cn/ubuntu bionic-security InRelease            
命中:7 https://mirrors.ustc.edu.cn/ubuntu bionic-proposed InRelease            
忽略:8 https://developer.download.nvidia.cn/compute/cuda/repos/ubuntu1804/x86_64  InRelease
获取:9 https://developer.download.nvidia.cn/compute/cuda/repos/ubuntu1804/x86_64  Release [697 B]
获取:10 https://developer.download.nvidia.cn/compute/cuda/repos/ubuntu1804/x86_64  Release.gpg [836 B]
忽略:11 https://developer.download.nvidia.cn/compute/cuda/repos/ubuntu1804/x86_64  Packages
忽略:11 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  Packages
获取:11 https://developer.download.nvidia.cn/compute/cuda/repos/ubuntu1804/x86_64  Packages [334 kB]
错误:11 https://developer.download.nvidia.cn/compute/cuda/repos/ubuntu1804/x86_64  Packages
  Hash 校验和不符
  Hashes of expected file:
   - Filesize:334425 [weak]
   - SHA256:368427dc3efc0befcdbcfc3927728878cde0066efea00e53edf3f126775c8149
   - SHA1:c41f40486a99db57667aafbdf2625d9930b42067 [weak]
   - MD5Sum:39e1a1818947e57e167e4f8fd4916e0b [weak]
  Hashes of received file:
   - SHA256:b8bba790dcb16497a66985c32b60a7bf85c2d3b731bd7fbd74d0941361f6f3cb
   - SHA1:acddf54609df69be3d3d07696a04f69f68b4b604 [weak]
   - MD5Sum:14d867b7f98d03ce7025a2ca2a32ca4a [weak]
   - Filesize:32 [weak]
  Last modification reported: Mon, 12 Oct 2020 12:33:34 +0000
  Release file created at: Fri, 09 Oct 2020 01:21:10 +0000
已下载 868 B，耗时 1分 37秒 (8 B/s)
正在读取软件包列表... 完成
E: 无法下载 https://developer.download.nvidia.cn/compute/cuda/repos/ubuntu1804/x86_64/Packages.gz  Hash 校验和不符
   Hashes of expected file:
    - Filesize:334425 [weak]
    - SHA256:368427dc3efc0befcdbcfc3927728878cde0066efea00e53edf3f126775c8149
    - SHA1:c41f40486a99db57667aafbdf2625d9930b42067 [weak]
    - MD5Sum:39e1a1818947e57e167e4f8fd4916e0b [weak]
   Hashes of received file:
    - SHA256:b8bba790dcb16497a66985c32b60a7bf85c2d3b731bd7fbd74d0941361f6f3cb
    - SHA1:acddf54609df69be3d3d07696a04f69f68b4b604 [weak]
    - MD5Sum:14d867b7f98d03ce7025a2ca2a32ca4a [weak]
    - Filesize:32 [weak]
   Last modification reported: Mon, 12 Oct 2020 12:33:34 +0000
   Release file created at: Fri, 09 Oct 2020 01:21:10 +0000
E: 部分索引文件下载失败。如果忽略它们，那将转而使用旧的索引文件。
ubuntu1804@ubuntu1804-VirtualBox:~$ cd AutowareArchitectureProposal/
ubuntu1804@ubuntu1804-VirtualBox:~/AutowareArchitectureProposal$ ./setup_ubuntu18.04.sh 
>  Do you run setup? This will take a while. [y/N] y
SUDO password: 
 [WARNING]: Unable to parse /home/ubuntu1804/AutowareArchitectureProposal/inventories/local-dev.ini as an inventory source

Do you install GPU modules(cuda: 10.2, cudnn: 7.6.5, TensorRT: 7.0.0)? (y/n) [y]: y

PLAY [localhost] *************************************************************************************************************************************

TASK [Gathering Facts] *******************************************************************************************************************************
ok: [localhost]

TASK [pause] *****************************************************************************************************************************************
skipping: [localhost]

TASK [cuda : CUDA (add nvidia key)] ******************************************************************************************************************
ok: [localhost]

TASK [cuda : CUDA (install CUDA 10.2 repository)] ****************************************************************************************************
ok: [localhost]

TASK [cuda : CUDA (install CUDA 10.2)] ***************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Failed to update apt cache."}
	to retry, use: --limit @/home/ubuntu1804/AutowareArchitectureProposal/ansible/localhost-setup-ubuntu18.04-devpc.retry

PLAY RECAP *******************************************************************************************************************************************
localhost                  : ok=3    changed=0    unreachable=0    failed=1   

Complete 
ubuntu1804@ubuntu1804-VirtualBox:~/AutowareArchitectureProposal$ ./setup_ubuntu18.04.sh --limit @/home/ubuntu1804/AutowareArchitectureProposal/ansible/localhost-setup-ubuntu18.04-devpc.retry
>  Do you run setup? This will take a while. [y/N] y
SUDO password: 
 [WARNING]: Unable to parse /home/ubuntu1804/AutowareArchitectureProposal/inventories/local-dev.ini as an inventory source

Do you install GPU modules(cuda: 10.2, cudnn: 7.6.5, TensorRT: 7.0.0)? (y/n) [y]: y

PLAY [localhost] *************************************************************************************************************************************

TASK [Gathering Facts] *******************************************************************************************************************************
ok: [localhost]

TASK [pause] *****************************************************************************************************************************************
skipping: [localhost]

TASK [cuda : CUDA (add nvidia key)] ******************************************************************************************************************
ok: [localhost]

TASK [cuda : CUDA (install CUDA 10.2 repository)] ****************************************************************************************************
ok: [localhost]

TASK [cuda : CUDA (install CUDA 10.2)] ***************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Failed to update apt cache."}
	to retry, use: --limit @/home/ubuntu1804/AutowareArchitectureProposal/ansible/localhost-setup-ubuntu18.04-devpc.retry

PLAY RECAP *******************************************************************************************************************************************
localhost                  : ok=3    changed=0    unreachable=0    failed=1   

Complete 

ubuntu1804@ubuntu1804-VirtualBox:~/AutowareArchitectureProposal$ sudo apt update 
[sudo] ubuntu1804 的密码： 
命中:1 http://mirrors.ustc.edu.cn/ros/ubuntu bionic InRelease
命中:2 https://mirrors.ustc.edu.cn/ubuntu bionic InRelease                                                                                           
命中:3 https://mirrors.ustc.edu.cn/ubuntu bionic-updates InRelease                                                                                   
命中:4 http://linux.teamviewer.com/deb stable InRelease                                                                           
命中:5 https://mirrors.ustc.edu.cn/ubuntu bionic-backports InRelease                                                              
命中:6 https://mirrors.ustc.edu.cn/ubuntu bionic-security InRelease                                          
命中:7 https://mirrors.ustc.edu.cn/ubuntu bionic-proposed InRelease                                         
错误:8 https://developer.download.nvidia.cn/compute/cuda/repos/ubuntu1804/x86_64  InRelease                 
  连接失败 [IP: 58.215.177.19 443]
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
有 61 个软件包可以升级。请执行 ‘apt list --upgradable’ 来查看它们。
W: 无法下载 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/InRelease  连接失败 [IP: 58.215.177.19 443]
W: 部分索引文件下载失败。如果忽略它们，那将转而使用旧的索引文件。
~
~
~
~
This is the result of the virtual machine, in fact, the error report on the physical machine is the same.

I have tried it for almost a week. During this period, I also tried to use the command line proxy: export https_proxy='http://127.0.0.1:41473/'       export http_proxy='http://127.0.0.1:41473/'

The final result is the same as above

Please help me, thank you very much :)

LalithVipulananthan, 2020-10-13:
@0-0-07 Thank you for reporting this installation issue! 

Googling the original Ansible error message turned up [this Ansible issue](https://github.com/ansible/ansible/issues/30754) which suggests running `apt-get install apt-transport-https` will resolve the problem.

If the problem continues to occur then I recommend requestin assistance at [Ask Ubuntu](https://askubuntu.com/), since the issue seems to be caused by the `apt update` command failing, and is not related to the setup script nor any issues with the AutowareArchitectureProposal code itself.

0-0-07, 2020-10-13:
> @0-0-07 Thank you for reporting this installation issue!
> 
> From what I can see, the problem appears to be occurring due to a failure to update the CUDA package. Could you try running `sudo rm /var/lib/apt/lists/* -vf` first, and then run `sudo apt update` again please?

@LalithVipulananthan  🥇 👍  Thank you for your precious time, the method you provided is useful to me.

closed 2020-10-13 by 0-0-07