tier4/AutowareArchitectureProposal.proj #77: [ROS2 Porting] Meeting 2020-10-15

mitsudome-r, 2020-10-14:
1. Go over actions from previous meeting:
  - https://github.com/tier4/AutowcreArchitectureProposal/issues/72
2. Report about any blockers
  - Use of timer has been updated https://github.com/tier4/AutowareArchitectureProposal/blob/ros2/docs/porting-to-ROS2.md#replicating-rostimer
  - Use of tf_listeners with timeout (tf2::BufferCore vs tf2::BufferClient) https://github.com/tier4/Pilot.Auto/issues/23

fred-apex-ai, 2020-10-14:
@mitsudome-r Because these issues come up repeatedly during reviews, I want to propose to agree in the meeting on how to uniformly handle

## composition
port existing main vs registering components as in https://index.ros.org/doc/ros2/Tutorials/Composition/

## ament_auto
shorter cmake files, less repetition vs keeping things mostly as they are

@nnmm perhaps you come up with more topics?

nnmm, 2020-10-14:
Yes, it would be great to have a written list of things that we discussed to do now vs later. My current understanding is

### Now
* Make each node compile and launch without issues (obviously)
* Use ament_auto and stricter warning flags (recommended by Chris/Dejan in Slack)
* Remove superfluous dependencies from package.xml (hasn't been discussed yet)
* Fix code style issues in the lines/parts that are changed anyway (hasn't been discussed yet, but clear win)

### Later (but before the AutowareAuto integration)
* Update README.md files (hasn't been discussed yet
* Convert to components (I think this was decided to be a not-to-do for now)
* Create custom messages to replace the deprecated std_msgs::Float32 etc. (hasn't been discussed yet)
* Rename files, e.g. `_core` files to `_node`, `.h` to `.hpp` (hasn't been discussed yet)

mitsudome-r, 2020-10-15:
## Adding Linter
* Adding lint to packages is quite easy. https://github.com/tier4/Pilot.Auto/pull/30
* However, fixing the lint issues might require many changes on source code that would be outside of scope, which might make review difficult at this stage.
* my suggestion is to not to add lint check for now. We can add lints to all packages after porting is done.

jilaada, 2020-10-15:
### `multi_object_tracker` blockers

Waiting for port of unique_identifier package to ROS2/Foxy: https://github.com/ros2/unique_identifier_msgs/issues/4#issuecomment-708932079

fred-apex-ai, 2020-10-15:
> Waiting for port of unique_identifier
https://lowrey.me/guid-generation-in-c-11/

jilaada, 2020-10-15:
> https://lowrey.me/guid-generation-in-c-11/

@fred-apex-ai Ah, this is what I was referring to in the meeting earlier. I can implement this for the time being while the UUID package is being ported. (The UUID ROS message itself is already ported so I think this should serve has an immediate solution to the blocker - I just wasn't sure if there were any design decisions around using the ROS UUID library vs the C++11 standard library)

fred-apex-ai, 2020-10-15:
>  I just wasn't sure if there were any design decisions around using the ROS UUID library vs the C++11 standard library)
Of course I wasn't there but I would think I'd rather use a well tested external package than Implement it myself and possibly make subtle mistakes. @jilaada

closed 2020-10-21 by mitsudome-r