tier4/AutowareArchitectureProposal.proj #84: [ROS2 Porting] Meeting 2020-10-16

mitsudome-r, 2020-10-16:
1. Go over action items from previous meeting https://github.com/tier4/AutowareArchitectureProposal/issues/77
2. Go over any blockers (Please report any blockers as a comment)

fred-apex-ai, 2020-10-17:
@mitsudome-r I suppose this issue is for 2020-10-19, please correct the issue title. I won't be able to attend the meeting but here is my status update:

- was blocked by osqp. After lots of back and forth, it's now added as a subtree vendor pkg in https://github.com/fred-apex-ai/autoware.iv.universe/pull/1
- so now osqp_interface is ready to be merged https://github.com/tier4/Pilot.Auto/pull/14 Its history is now a big mess. I squashed in a [new branch](https://github.com/fred-apex-ai/autoware.iv.universe/tree/port-osqp-interface-squashed) and tada, `colcon` doesn't find the package `osqp_interface` anymore. I can't spot any differences in the `osqp_interface/` directory. I don't like `colcon`. If the messy history is ok, then please go ahead and merge the PR
- next step: can now continue porting mpc_follower. It's half way done, major outstanding work is to port dynamic reconfigure for parameters

simon-t4, 2020-10-19:
pointcloud_preprocessor packages - not blockers but an update
- working on cocatenate_data_nodelet/component
- inherits NodeletLazy nodelet_topic_tools - but only seems to have typical sub/pubs
-  reads a variable amount of topic names from parameter list using an XmlRpc library structure (allows for an array of parameters)
- is this structure still present in ROS2 paramters? (set up in sensing launch file - should move to xml parameter file which is passed to node?~

TakaHoribe, 2020-10-19:
> OpenCV dependency

Just adding `libopencv-dev` works well. Thanks.

dejanpan, 2020-10-19:
@dejanpan TODOs:

1. organize colcon tutorial
2. address: https://gitlab.apex.ai/ApexAI/grand_central/-/issues/8423
3. How to deal with the dependency on openCV => https://github.com/tier4/AutowareArchitectureProposal/issues/84 -> clarify both approaches
4. How to deal with CUDA, TensorRT dependency
5. Nodelet to component migration specifics: https://github.com/tier4/AutowareArchitectureProposal/issues/84#issuecomment-711673364 
6. Covariances are not transformed in TF2: 
    1. https://github.com/nnmm/autoware.iv.universe/blob/86199d74595fba38e5c50f824f94385e7cb2df23/localization/pose_estimator/ndt_scan_matcher/ndt_scan_matcher/src/ndt_scan_matcher_core.cpp#L193
   1. https://github.com/ros2/geometry2/pull/312

dejanpan, 2020-10-21:
@simon-t4 regarding this:
> * reads a variable amount of topic names from parameter list using an XmlRpc library structure (allows for an array of parameters)
> * is this structure still present in ROS2 paramters? (set up in sensing launch file - should move to xml parameter file which is passed to node?~

If you can next time, give me a more concrete question like this:
1. The question is about XmlRpc::XmlRpcValue: https://github.com/tier4/Pilot.Auto/blob/master/sensing/preprocessor/pointcloud/pointcloud_preprocessor/include/pointcloud_preprocessor/concatenate_data/concatenate_data_nodelet.h#L127
2. Which is being used like this: https://github.com/tier4/Pilot.Auto/blob/master/sensing/preprocessor/pointcloud/pointcloud_preprocessor/src/concatenate_data/concatenate_data_nodelet.cpp#L106-L124

How to imitate that in ROS 2?

Also, where is the: `set up in sensing launch file`?

dejanpan, 2020-10-21:
@simon-t4 to answer your question:
1. `XmlRpc::XmlRpcValue` is not available in ROS 2
2. To replicate a similar behavior in ROS 2 I would suggest to copy this approach:
    1. Parameter specification: https://github.com/ros-planning/navigation2/blob/d3fc28af28df764a153c85434f8b30270e963622/nav2_costmap_2d/launch/example_params.yaml#L32
    1. Parameter reading in: https://github.com/ros-planning/navigation2/blob/d3fc28af28df764a153c85434f8b30270e963622/nav2_costmap_2d/src/costmap_2d_ros.cpp#L91
    1. Parameter parsing: https://github.com/ros-planning/navigation2/blob/d3fc28af28df764a153c85434f8b30270e963622/nav2_costmap_2d/src/costmap_2d_ros.cpp#L178-L179
    1. makeFootprintFromString: https://github.com/ros-planning/navigation2/blob/d3fc28af28df764a153c85434f8b30270e963622/nav2_costmap_2d/src/footprint.cpp#L173

dejanpan, 2020-10-21:
## Colcon/Ament Tutorial Topics
1. how imported targets work, and how ament_auto works for example.
1. I would like to know a detailed example of how ament and colcon work. There are many simple file examples, but I still don't have a detailed understanding of how they work.
1.  package lookup and finding a package or not, inconsistent or no error mesasge
1. when colcon takes an installed package or when it decides it needs to rebuild
1. consistent build options: doesn't know how packages where built, so can mix debug and release builds freely, whatever is in install/
1. ament_auto has little to no documention built, I just look into the source code on github
1. ament_export and transitive dependencies
1.  sourcing setup.bash etc, this bit me multiple times with inconsistencies
1. I had created two issues https://github.com/colcon/colcon-core/issues/399#issuecomment-710065767 and https://github.com/colcon/colcon-cmake/issues/96#event-3894910528 describing more details

wjwwood, 2020-10-21:
@dejanpan your comment here: https://github.com/tier4/AutowareArchitectureProposal/issues/84#issuecomment-713882022 lgtm

closed 2020-10-22 by mitsudome-r