tier4/AutowareArchitectureProposal.proj #86: Hardware Problem

0-0-07, 2020-10-17:
@tier4
Excuse me, have you considered selling all the hardware used in this software in an international Amazon store?

Because there is no IMU with the same model as in the software in our national electronic mall

If it is really possible, please tell me, is the item listed on Amazon URL？

sincere thanks:)... (*^ω^*).........(´▽｀)ノ♪

LalithVipulananthan, 2020-10-19:
@0-0-07 Given that Tier IV is a provider of reference designs and services for autonomous driving based on Autoware, rather than a hardware provider, it is unlikely that we will start selling the hardware used for this Autoware Architecture Proposal. However, things may change in the future, so thank you for the kind suggestion. 

In terms of purchasing the IMU mentioned in this proposal, it appears that the sensor can only be bought directly from the manufacturer, or perhaps from specialized vendors. That said, the manufacturer (Tamagawa) accepts purchase enquiries in English as well as Japanese, so I've included a link to the [English contact form](
http://www.mems.tamagawa-seiki.com/en/contact/form/).

0-0-07, 2020-10-19:
@LalithVipulananthan  Thanks for your reply,
 I will browse the link you provided 
 *\(^o^)/*

closed 2020-10-19 by 0-0-07