tier4/AutowareArchitectureProposal.proj #91: [ROS2 Porting] Meeting 2020/10/22

mitsudome-r, 2020-10-21:
1. Go over action items from the last meeting: https://github.com/tier4/AutowareArchitectureProposal/issues/84
1. Go over any blockers
  - Setting up CUDA
  - Add message type for debug topic or use examle_msgs?

nnmm, 2020-10-21:
We could try to reach a decision on global parameter server node vs local parameters.

jilaada, 2020-10-22:
## Blockers:
My blockers issues are localized rather than systemic in this project so feel free to skip the discussion in the meeting if time does not permit. I do have one issue I would like to discuss: an effort to create smaller more easily reviewed PRs. For larger packages such as `system_monitor`, many independent executables are created, would it be wiser to create multiple, easier to create and review PRs at semi-complete stages of the port? 

### autoware_error_monitor
https://github.com/tier4/Pilot.Auto/pull/42
* launch files also launch the `diagnostics_aggregator` however parameter loading causes issues (doesn't seem to properly load the analysers): struggling to find examples in ROS2. Potentially some key changes?
* dependency to `dummy_diag_publisher` not specified in the package.xml: this package is set to medium priority, should this be ignored for now since it is only a launch file dependency?

### autoware_state_monitor
* use of `topic_tools::ShapeShifter` which hasn't been ported to ROS2 yet. Potential work-around, needs investigation.

fred-apex-ai, 2020-10-22:
## Reach decision

1. vcs proj file vs git subtree

## blockers
osqp_interface still not merged https://github.com/tier4/Pilot.Auto/pull/45

## progress
finally opened PR for mpc_follower

dejanpan, 2020-10-22:
@nnmm @fred-apex-ai @jilaada @simon-t4 @esteve @TakaHoribe @mitsudome-r @nik-tier4 I unfortunately fell sick to food poisoning and couldn't attend the meeting today. I did most of my todos here https://github.com/tier4/AutowareArchitectureProposal/issues/84#issuecomment-712390875.

@simon-t4 
Re LazyNodelets - they are not available as 1-1 in Ros2 because topic_tools is not ported yet. I am still investigating what the proper way of porting is.

@nnmm for the transform of covariances, please comment on the PR from Josh to get it back ported to Foxy. @mention whoever approved that PR.

TakaHoribe, 2020-10-22:
> @TakaHoribe create autoware_debug_msgs package to contain deprecated msgs with timestamp to be used for debug topics

I noticed the std_msgs are used not only for the debugging, for example max_velocity, engage, emergency. So it's better to name the autoware_std_msgs. I think.

closed 2020-10-26 by mitsudome-r