tier4/AutowareArchitectureProposal.proj #92: Replace the deprecated std_msgs

TakaHoribe, 2020-10-21:
As discussed here https://github.com/tier4/AutowareArchitectureProposal/issues/77#issue-721207488, we need to create a new message type to replace the deprecated `std_msgs`.

`ColorRGBA`, `Empty` and `Header` are still alive in the `std_msgs`.

Other message types used are below.

 - Int32
 - Float32
 - Float32MultiArray
 - Float64
 - Float64MultiArray
 - Bool
 - String

In the meantime, I'm going to define these types with a timestamp added as `autoware_util_msgs` or `autoware_std_msgs`.
Do you have any other suggestions?

kenji-miyake, 2020-10-21:
Since I think these should be used only for debugging, I personally prefer autoware_debug_msgs.
Or if adding timestamp isn't necessary, not creating autoware_*msgs for these and just using [example_interfaces](https://github.com/ros2/example_interfaces) is fine.

nnmm, 2020-10-21:
I would like to better understand the motivation for having these messages instead of specific message types like ROS2 generally recommends.

As I see it, developers might want to use use debug messages either
* when they are actually debugging (something is broken)
* when they want to provide a way to measure performance or to inspect the node (long-term)

For actual debugging, it's easiest to use primitive type like Float32. I agree that you could use example_interfaces or even std_msgs there. But since you only need it temporarily to fix a bug, I think these publishers should not be merged into the main branch.

When a package wants to provide a topic in the long term, I would assume that the recommendation to define custom messages applies. That is what I did in https://github.com/tier4/Pilot.Auto/pull/48 for example. Do you think the debug message format needs to change too often for such a message to make sense? I assumed that it doesn't. Additionally I find it nice to have a single "debug_info" topic instead of many individual topics where it's not clear from their name that they're just for debugging.

TakaHoribe, 2020-10-21:
@nnmm The content of the debug topic changes at least depending on the algorithm implementation. 
And these algorithms are still under development and not far from perfect, so it will be changed frequently. At least it does in TierIV for the frequently developed packages.

If we change the message type, we have to think about the backward-compatibility; if you find some other useful information and put it in the debug topic, you have to convert all rosbags ever for the type change. And it's not an easy task. This is the main reason why we use a primitive message type.

So we should define the message types only which you can be assured will not change completely (at least for the time being).

nnmm, 2020-10-21:
Thanks @TakaHoribe, I now have a better understanding of the motivation.

If ros2 bag could play back messages with unknown type, this would not be a problem since the debug messages are only published, never consumed. But that is [not possible yet](https://github.com/ros2/rosbag2/issues/307).

TakaHoribe, 2020-10-21:
Yes, as you say, if the rosbag and pub/sub system become more flexible for the message type, the problem will be solved.

mitsudome-r, 2020-10-22:
Within this meeting, we've decided to create autoware_debug_msgs inside Pilot.Auto.
I will keep this open until the packages is actually merged into the repository.

closed 2020-10-23 by mitsudome-r