tier4/AutowareArchitectureProposal.proj #96: [ROS2 Porting] 2020-10-26 Meeting

mitsudome-r, 2020-10-23:


fred-apex-ai, 2020-10-26:
## blockers
  - [PR](https://github.com/tier4/AutowareArchitectureProposal/pull/95) to install osqp only once should be merged

  ## in progress
  - had 90 min screen sharing session with @TakaHoribe to review his PR on velocity controller and my PR on mpc_follower more effectively. 
  - mpc_follower builds and runs: just need to test I can set every parameter with `ros2 param` and the updated value is used in the mpc
  - adding lessons learned to update porting-doc. Will new open PR today
  - @nnmm and myself looked into exception that @TakaHoribe reported but couldn't reproduce in similar situation in our  code. Will check out it his code directly

  ## done
  - osqp subtree removed
  - osqp_interface merged to pilot.auto

dejanpan, 2020-10-27:
@mitsudome-r I am pretty sure that you made longer meeting minutes for this meeting. Did you forget to hit save?

I captured 2 action items for myself:
1. [ ] figure out the state of porting for `topic_tools` package
1. [ ] add CUDA and TensorRT (From https://github.com/tier4/Pilot.Auto/pull/53) dependencies into the  https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/blob/master/.aderc-amd64-foxy

dejanpan, 2020-10-27:
@mitsudome-r couple of item for the next meeting:
1. Can Apex.AI self review the MRs to speed up the reviews a bit?
1. Can we move the meeting time 1h back so it is back at 9AM CET? This will be 5PM in JST

closed 2020-11-02 by mitsudome-r