tier4/AutowareArchitectureProposal.proj #103: The usage of the waitForTransform in ROS2-foxy

TakaHoribe, 2020-10-30:
Following the discussion in https://github.com/tier4/AutowareArchitectureProposal/pull/88, I've tested with a simple package (referring to the @nnmm's). 
https://github.com/TakaHoribe/tf2_example/tree/test-tf2-LookupException

But I still have an issue about the usage of the `waitForTransform()`.

When I run the following command for the package,
```
// terminal A
$ ros2 run tf_example waiter_case2 --ros-args -p timeout_ms:=1000

// terminal B
ros2 run tf_example publisher_case2
```

I got this.

```
horibe@ade:~/AutowareArchitectureProposal (ros2 *$% u=)$ ros2 run tf_example waiter_case2 --ros-args -p timeout_ms:=1000
[INFO] [1604032319.361292436] [minimal_subscriber]: timeout is 1000ms
[INFO] [1604032320.361729479] [minimal_subscriber]: tf is not ready. waiting another 1.000000 seconds for a->b transform
[INFO] [1604032321.362212528] [minimal_subscriber]: tf is not ready. waiting another 1.000000 seconds for a->b transform
[INFO] [1604032322.362829775] [minimal_subscriber]: tf is not ready. waiting another 1.000000 seconds for a->b transform
[INFO] [1604032323.149732191] [minimal_subscriber]: tf gets ready. try tf_future.get()...
[WARN] [1604032323.149826324] [minimal_subscriber]: catch LookupException. what:[Failed to transform from b to a]. try again.
[INFO] [1604032323.149856336] [minimal_subscriber]: tf gets ready. try tf_future.get()...
[INFO] [1604032323.149877946] [minimal_subscriber]: Successed to get transform. End processing.
```

This means, the first time we do `tf_future.get()` , it throws the `LookupException` even though the tf is ready.

Note: the way of the `mpc_follower` package doesn't throw the exception. It uses the `waitForTransform()` to just check if the tf is available [here](https://github.com/tier4/Pilot.Auto/blob/4d14dfe14c418b8e23e603567f182fb3c8a0b587/control/mpc_follower/src/mpc_follower_core.cpp#L1000-L1019), and use the `lookupTransform()` [here](https://github.com/tier4/Pilot.Auto/blob/4d14dfe14c418b8e23e603567f182fb3c8a0b587/control/mpc_follower/src/mpc_follower_core.cpp#L874-L893) after the tf future status is ready.

nnmm, 2020-10-30:
That is very interesting, so it's even more complicated than I thought. Maybe it has something to do with requesting `tf2::timePointZero`? I can test this a bit later in more detail.

A workaround to try could be to use the callback approach instead of the returned future.

fred-apex-ai, 2020-10-31:
@TakaHoribe @nnmm Now I have played around with various settings. I feel that if we just want to block in the constructor until the transform is available, the asynchronicity of `waitForTransform` is just overkill and slight variations lead to hanging etc. How about something as simple as this which seems easier to understand? https://github.com/fred-apex-ai/tf2_example/blob/test-tf2-LookupException/src/waiter_case2.cpp#L77

I can confirm that `tf` fails on the first lookup even if the transform should be available. perhaps this is related to fillling the local buffer? And I played around and found out that you need to give it some time so the 2nd lookup succeeds. On my machine, 3 ms was not enough but 30 ms is. It's awkward and not robust. We should ask elsewhere why that is. Looking into https://github.com/ros/geometry2/blob/noetic-devel/tf2/src/buffer_core.cpp where the heavy lifting is done, I couldn't see why. I was suspecting that there is a map which has to be filled by a lookup but that seems to be called only in `setTransform`

fred-apex-ai, 2020-11-01:
All of this may just be so weird because in this little test everything happens in the init phase, and the publisher might publish before/during/after we check for the transform availability. In my test, it seems there is no problem if I just wait longer than the publisher period before  calling `waitForTransform`

TakaHoribe, 2020-11-02:
> All of this may just be so weird because in this little test everything happens in the init phase

I moved the function into timer_callback, but still the same problem happens.

> it seems there is no problem if I just wait longer than the publisher period before calling waitForTransform

Even if I wait a long time (1s) before `waitForTransform()`, sometimes the exception happens.

This exception happens only the first time when the tf is published. If the tf is already published when I launch the waiter, there is no exception.
It seems the waitForTransform has a complicated process to check for the latest tf (`tf2::TimePointZero`). We can use Fred's simple code and push this question to ros/geometry2.

mitsudome-r, 2020-11-02:
@cvasfi could you look into this?

cvasfi, 2020-11-02:
From reading the source code, I understand that the mistake here is to assume that `waitForTransform` supports the special `tf2::TimePointZero` input. This specific timepoint is treated specially in `canTransform` and `lookupTransform` but not in the [testTransformableRequests](https://github.com/ros2/geometry2/blob/foxy/tf2/src/buffer_core.cpp#L1474-L1478) function where it is treated as any other query. This results in `BufferCore` thinking that the query time is simply too old to exist. Hence the error.

I haven't dived too deep but i think the reason it only fails on the first iteration despite having data is that after the `TransformableRequest` is called for the first time and the provided lambda is executed, it is simply removed from the queue, meaning that the logic to result in an error will only happen again once we give it another transformable request.

_______________________________________________________________________________

@fred-apex-ai  

> I can confirm that `tf` fails on the first lookup even if the transform should be available.

It fails on the first lookup with 0ms timeout because the tf2_listener hasn't had a chance to receive and insert the transform yet so I think in the following case, everything works as intended:

```cpp
        while(!tf_buffer_.canTransform(input, output, tf2::TimePointZero, 0ms, &msg) && rclcpp::ok()) {
            RCLCPP_INFO(get_logger(), tf_buffer_.allFramesAsString());
            RCLCPP_INFO(
                    get_logger(), "waiting %d ms for %s->%s transform to become available",
                    timeout_ms_, input, output);
            rclcpp::sleep_for(std::chrono::milliseconds(timeout_ms_));
        }
        RCLCPP_INFO(get_logger(), "transform available");
```

I added a log statement to print all of the existing nodes within the buffer. On each iteration, we see that can transform failed and there is actually no data in the buffer yet.

___________________________________________________________________________

Conclusion: If `TimePointZero` is desired to be used, I am also onboard with Fred's solution.

fred-apex-ai, 2020-11-02:
@nnmm and I discussed this on the whiteboard. There are essentially two use cases

## node ready?
The goal is to ensure that node can do its job properly and that the transforms it needs are generally available.
There are two options we have
1. wait indefinitely in the constructor
2. check locally where the tf is to be used if it's available. None of that requires `waitForTransform`

I take over the responsibility and write down the recomendation in a separate PR

## getting the right transform
For example if latencies in the system are an issue and one node has to wait for the transform to arrive for a particular timestamp to best transform a data set

This part still requires some investigation and I leave it to @nnmm to carry out the investigation and report the conclusion in this porting guidelines in a separate PR

[image]

fred-apex-ai, 2020-11-03:
@TakaHoribe I went for option B on the left of the image: it's not necessary to get into the complications of `waitForTransform` just to check that there was some transform available. It's enough to do catch around a `lookupTransform` and then return early. Only downside is that an error message is output in the console every time this fails. I do this here https://github.com/tier4/Pilot.Auto/pull/52/files?file-filters%5B%5D=.cfg&file-filters%5B%5D=.cpp&file-filters%5B%5D=.py&file-filters%5B%5D=.test&file-filters%5B%5D=.txt&file-filters%5B%5D=.yaml&file-filters%5B%5D=No+extension#diff-30dcb55e06b0d11249fb5d74414c4c96ceb4b891a18142afed00452ed9ef2ec3R873

nnmm, 2020-11-03:
@mitsudome-r @TakaHoribe @jilaada Today I tried testing `waitForTransform` a bit further – there are basically three possible scenarios that should work as expected:
* Calling `waitForTransform` when the transform is already available (should return the transform immediately)
* Calling `waitForTransform` when the transform arrives while waiting (should also not throw an error)
* Calling `waitForTransform` when the transform doesn't arrive before the timeout expires (should signal failure)

In Taka's case, the first iteration always failed and subsequent ones did not. I was able to reproduce that scenario 2 (= first iteration) always throws an exception when you request `tf2::TimePointZero`, while scenario 1 works fine. I had previously only tested requests with times other than `tf2::TimePointZero` and didn't test for scenario 1 at all. I now tried out as many variations of calling `waitForTransform` as possible, here are the results: https://docs.google.com/spreadsheets/d/1BvFIMwp0kSkQecw2dkkyj4kR9edFLRDDBBTGynHBAN8/edit#gid=0

I am not 100% certain in these results because one case contradicts what I had found out earlier and want to write simpler tests to reproduce them. However, there seems to be no way of using it that works as expected, at least for single-threaded executors. I would therefore suggest I create a more easily reproducible test case, and open an issue.

### What we can do
Since `waitForTransform` seems to be very difficult to use or buggy, I would suggest we ask ourselves if our code really needs it. To expand on what Fred wrote above, many nodes have a subscription callback that checks if auxiliary data is there and returns early from the callback if it isn't. We can treat transforms the same way as this auxiliary data, i.e. just do a simple `lookupTransform` with no timeout and return early if it fails.

TakaHoribe, 2020-11-04:
@nnmm @fred-apex-ai @cvasfi Thanks a lot. So using the `waitForTransform` is not appropriate at least for now.

In the control module (`mpc_follower` or `velocity_controller`), it just needs the latest TF, so let's go with the simple try-catch as in mpc.

```
geometry_msgs::msg::TransformStamped transform;
try {
  transform = tf_buffer_.lookupTransform("map", "base_link", rclcpp::Time(0));
} catch (tf2::TransformException & ex) {
  RCLCPP_WARN_SKIPFIRST_THROTTLE(
    get_logger(), *get_clock(), (5000ms).count(),
    "cannot get map to base_link transform. %s", ex.what());
  return;
}
```

@cvasfi I'd like to know for my understanding, is the unsupported `timeZero` in `waitForTransform` intensional? Or just not implemented yet? I don't know the design concept of the TF, so please let me know if you know something.

TakaHoribe, 2020-11-04:
Note: the code above shows warning `Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist` every time the function is called until the tf is ready.

cvasfi, 2020-11-04:
@TakaHoribe 
>  I'd like to know for my understanding, is the unsupported `timeZero` in `waitForTransform` intensional? Or just not implemented yet? I don't know the design concept of the TF, so please let me know if you know something.

I think it might be intentional. One argument for not implementing the wait for latest transform is that, once the transform between is initialized once, then you will always have a *last* transform and then waiting won't make sense. At that point it makes zero sense to call `waitForTransform` and run all that async. logic whereas one could just call `lookupTransform` with `TimePointZero` and get the same behavior free from threading boilerplate.

In my opinion, `waitForTransform` is for data oriented lookup scenarios where an action to be triggered waits for a specific transform to become available. (e.g. I get a point cloud i want to work with and suppose i can't process it until i know where it is in the map so i wait for `map-bl` transform at the timestamp of the message.)

TakaHoribe, 2020-11-19:
A quick summary

Question: "the first time we do `tf_future.get()`, it throws the `LookupException` even if the `tf_future` says it is ready."

Answer: "`waitForTransform()` doesn't support the `tf2::TimePointZero`"

closed 2020-11-19 by TakaHoribe

nnmm, 2020-11-19:
See also https://github.com/ros2/geometry2/issues/348