tier4/AutowareArchitectureProposal.proj #106: Require C++ standard strictly

fred-apex-ai, 2020-11-03:
# Problem

We currently write in a package's `CMakeLists.txt`

```cmake
```cmake
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()
```

and cmake then invokes `g++ -std=gnu++14`. But we shouldn't allow GNU extensions because it is not portable and with the above, we are not guaranteed we really get C++14, it could also be C++11 as a fallback; see https://crascit.com/2015/03/28/enabling-cxx11-in-cmake/

# Proposed solution

All ported packages should instead use

```cmake
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
  set(CMAKE_CXX_STANDARD_REQUIRED ON)
  set(CMAKE_CXX_EXTENSIONS OFF)
endif()
```

esteve, 2020-11-03:
Sounds good to me, but should we defer this for after porting? As part of all the other cleanups, I mean.

fred-apex-ai, 2020-11-03:
@esteve yes, that's what I had in mind. I updated the issue description accordingly

closed 2021-09-15 by mitsudome-r