tier4/AutowareArchitectureProposal.proj #107: difference between AutowareArchitectureProposal lidar_apollo_instance_segmentation and Autoware.AI lidar_apollo_cnn_seg_detect

peterlee909, 2020-11-04:
Is there any difference between AutowareArchitectureProposal lidar_apollo_instance_segmentation and Autoware.AI lidar_apollo_cnn_seg_detect?

In my opinion, the preprocessing and postprocessing of point cloud are the same between two packages.
Is the used model are better in AutowareArchitectureProposal lidar_apollo_instance_segmentation package?

Thanks.

yukkysaito, 2020-11-16:
lidar_apollo_instance_segmentation includes code refactoring and object orientation estimation.

Also, there are three types of DNN models: velodyne 16, 64, and 128 layers.

closed 2021-04-16 by LalithVipulananthan

yunqiushuiman, 2021-09-05:
@yukkysaito  Hi, could you please explain why need to add a factor 0.5 to estimate the object orientation like this: " obs->heading = std::atan2(vec_y, vec_x) * 0.5 "  in lidar_apollo_instance_segmentation package?  Thank you!