tier4/AutowareArchitectureProposal.proj #115: [ROS2 Porting] Meeting 2020/11/09

mitsudome-r, 2020-11-06:
1. Go over previous meeting https://github.com/tier4/AutowareArchitectureProposal/issues/108
2. Discuss about any blockers

fred-apex-ai, 2020-11-06:
## Question

Why are there two directories named `scenario_api`? One is a package, the other code seems to be unused. Remove it?

```
➜  AutowareArchitectureProposal git:(lessons-from-porting-mpc) ✗ tree ./src/simulator/scenario_runner/scenario_api
./src/simulator/scenario_runner/scenario_api
├── include
│   └── scenario_api_event.h
└── src
    └── scenario_api_event.cpp

2 directories, 2 files
➜  AutowareArchitectureProposal git:(lessons-from-porting-mpc) ✗ tree ./src/simulator/scenario_runner/api/scenario_api
./src/simulator/scenario_runner/api/scenario_api
├── CMakeLists.txt
├── COLCON_IGNORE
├── include
│   └── scenario_api
│       ├── scenario_api_coordinate_manager.h
│       ├── scenario_api_core.h
│       └── scenario_calc_dist_utils.h
├── launch
│   ├── scenario_api_sample.launch
│   └── scenario.launch
├── package.xml
├── rviz
│   └── planning_simulator.rviz
└── src
    ├── scenario_api_calc_dist_utils.cpp
    ├── scenario_api_coordinate_manager.cpp
    ├── scenario_api_core.cpp
    └── scenario_api_node.cpp

5 directories, 13 files
```

mitsudome-r, 2020-11-09:
@fred-apex-ai You can ignore the folder without package.xml.

dejanpan, 2020-11-09:
@mitsudome-r can you change `@dejan` => `@dejanpan` because with the former you are now tagging some other guy?

mitsudome-r, 2020-11-09:
Thanks for pointing that out. I've fixed the name.

closed 2020-11-16 by mitsudome-r