tier4/AutowareArchitectureProposal.proj #120: Run ./setup_ubuntu18.04.sh fail!!!

JackRen88, 2020-11-12:
Hello,everyone.
After running  the setup script `./setup_ubuntu18.04.sh`,there is a error that is
`fatal: [localhost]: FAILED! => {"changed": false, "failed": true, "msg": "Unable to install package: E:Could not open file http://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/cuda-repo-ubuntu1804_10.2.89-1_amd64.deb - open (2: No such file or directory)"}

PLAY RECAP *********************************************************************
localhost                  : ok=2    changed=0    unreachable=0    failed=1   

Complete 
`
How should I deal with it?

yukkysaito, 2020-11-16:
thank you for your comment.
It probably failed to download. Can you run it again and still fail?

LalithVipulananthan, 2021-04-16:
@JackRen88 @HuangCongQing Hi guys. Apologies for the delayed follow-up, but could you both confirm if this issue is still occurring with the v0.9.1 branch please? I've run that myself recently a few times for testing, and have experienced no issues with the CUDA installation step, so I don't think there's a problem with the script.

Just to confirm that you are able to access the NVidia download site, can you try opening a browser and pasting in the link:
http://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/cuda-repo-ubuntu1804_10.2.89-1_amd64.deb

If that works, then the script should also work.

ajay1606, 2021-05-03:
@LalithVipulananthan Thanks for your response, i am also getting the same error while running command ./setup_ubuntu18.04.sh
[image]

Also, I have tried to access the Nvidia download site one that you shared in the previous link. Please could you guide, where am missing ?

Kind regards,
Ajay

LalithVipulananthan, 2021-05-07:
@ajay1606 I've had a quick look at the screenshot, and though the error is occurring during the execution of the setup_ubuntu18.04.sh script, the error is different to the original one. Your error is occurring during the "update rosdep" step, but the OP's error related to CUDA installation.

Could you confirm the result of running the following command please?
`update-alternatives --list python`

ajay1606, 2021-05-07:
@LalithVipulananthan Thank you so much for your response. And have tried as you suggested, getting error as below:
[image]

Is there any other option to fix this /

Thanks

LalithVipulananthan, 2021-05-07:
@ajay1606 Well, the error seems to indicate there is a problem with python-rospkg, so we're going to have to try a few things to see what works. We've established that you don't have two versions of Python installed which is good. Now let's see if you have python-rospkg installed or not. What's the result of running the following command?

`apt list python-rospkg -a`

ajay1606, 2021-05-10:
@LalithVipulananthan thanks much for your assistance, and here is what i got after running command

 **apt list python-rospkg**

[image]

LalithVipulananthan, 2021-05-10:
@ajay1606 Thank you. So that output tells me that your installed python-rospkg package is 1.2.9-100 and not 1.3.0, which explains the errors that you've been getting. Please upgrade python-rospkg by running the following commands:
```
sudo apt-get update
sudo apt-get --only-upgrade install python-rospkg
```

After that has completed, please try rerunning the AutowareArchitectureProposal setup script. That should resolve the issue, but if not, then please let me know if any further errors occur.

ajay1606, 2021-05-10:
@LalithVipulananthan Excellent, and many thanks for your kind assistance, really appreciate your help. 

[image]

Finally, it working well.

Kind Regards,
Ajay

closed 2021-05-10 by LalithVipulananthan