tier4/AutowareArchitectureProposal.proj #133: [ROS2 Porting] Meeting 2020/11/30

mitsudome-r, 2020-11-30:
1. Go over previous meeting minutes.
2. Discuss current blockers

fred-apex-ai, 2020-11-30:
## parameters not available anymore in ros2
```
<!-- velodyne_driver/DriverNodelet -->
<param name="scan_phase" value="$(var scan_phase)"/>
<param name="sensor_timestamp" value="$(var sensor_timestamp)" />
```

## nodelet intention and missing parameters
also the cloud nodelet itself was turned into transform or convert node. Which one do we need?
```
<!-- velodyne_pointcloud/CloudNodelet -->
<param name="num_points_threshold" value="$(var num_points_threshold)"/>
<param name="invalid_intensity" value="$(var invalid_intensity)"/>
<param name="scan_phase" value="$(var scan_phase)"/>
```

## nodelet gone

```xml
<!-- gone from velodyne repo -->
<node pkg="nodelet" exec="nodelet" name="$(var manager)_fix_distortion" args="load velodyne_pointcloud/InterpolateNodelet $(var manager)">
    <remap from="velodyne_points_ex" to="mirror_cropped/pointcloud_ex" />
    <remap from="velodyne_points_interpolate" to="rectified/pointcloud" />
    <remap from="velodyne_points_interpolate_ex" to="rectified/pointcloud_ex" />
  </node>
```

nnmm, 2020-11-30:
# Ament linters
## How to use ament linters
See [this document](https://github.com/ament/ament_lint/blob/master/ament_lint_auto/doc/index.rst). The linters are executables wrapped by cmake packages.

## Results for the `pose_initializer` package
I added that to the `pose_inializer` package and ran
```bash
colcon build --packages-up-to pose_initializer
colcon test --packages-select pose_initializer
colcon test-result --verbose
```

### `ament_cmake_copyright`
It complains for all source files that it doesn't find a copyright notice, although there is one, because it [doesn't understand `/* */`](https://github.com/ament/ament_lint/blob/foxy/ament_copyright/doc/index.rst).
This can be fixed by running `ament_copyright --add-missing 'Autoware Foundation' apache2` (or whatever license holder is appropriate) and then deleting the existing license. That's probably not too hard to do in a batch operation as long as the existing licenses all look the same.

### `ament_cppcheck`
It prints

```
- cppcheck
  <<< failure message
    -- run_test.py: invoking following command in '/home/nikolai.morin/t4/src/autoware/pilot.auto/localization/util/pose_initializer':
     - /opt/ros/foxy/bin/ament_cppcheck --xunit-file /home/nikolai.morin/t4/build/pose_initializer/test_results/pose_initializer/cppcheck.xunit.xml --include_dirs /home/nikolai.morin/t4/src/autoware/pilot.auto/localization/util/pose_initializer/include
    [include/pose_initializer/pose_initializer_core.h:33]: (error: syntaxError) Code 'classPoseInitializer:' is invalid C code. Use --std or --language to configure the language.
    1 errors
    -- run_test.py: return code 1
    -- run_test.py: verify result file '/home/nikolai.morin/t4/build/pose_initializer/test_results/pose_initializer/cppcheck.xunit.xml'
  >>>
```
This is not very readable – there is a lot of irrelevant stuff in there. The cause of the error is that the file is assumed to be `C` and not `C++` because of the file name extension. This can easily be fixed in a batch rename operation.

### `ament_cpplint`
This one has a lot of output, so I split it into error categories:

#### Requiring header guards instead of `#pragma once`
I didn't find a tool to generate those automatically, so it would potentially mean a moderate amount of work for very little gain, at least where `#pragma once` is already there.

#### Ordering of `#include`s
The `pose_initializer_core.hpp` has the following includes:

```cpp
#include <memory>
#include <string>

#include <rclcpp/rclcpp.hpp>
// more includes
```
On the `rclcpp` line, `cpplint` outputs this very misleading message: `Found C system header after C++ system header. Should be: pose_initializer_core.h, c system, c++ system, other.`
Here's what's wrong with that:
* there is no `pose_initializer_core.h`, we're already in the header file
* `rclcpp/rclcpp.hpp` is not a C system header. `cpplint` thinks angle brackets = system header.
* I don't think the order should be like that: If we include system headers _after_ the headers of other Autoware/in-house packages, we can guard a bit better against missing includes in those. See [this article](https://blog.knatten.org/2010/07/01/the-order-of-include-directives-matter/) for the rationale. It is a very frequent problem in the codebase that there are superfluous and missing headers.

Changing angle brackets of `.h` or `.hpp` includes to quotes should be not too much work, given that C system headers are rarely used in the codebase. But reordering the includes is more work.

#### Include what you use
Tells you to add missing C++ system headers. A moderate amount of work, because it is manual and probably affects most packages, but done in a few minutes per package. It doesn't bring that much of a quality improvement because if a build breaks because of this, it will be pretty obvious and easy to fix.

#### TODOs
There are a bunch of `Missing username in TODO; it should look like "// TODO(my_username): Stuff.` – a very good check in my opinion, since the code base has many `// TODO` comments that do not even explain what the TODO is. When the code base transitions to AutowareAuto, it should be self-contained and not rely on access to the original authors.

Fixing that check would probably require pinging a lot of the original authors. Before that, we should clarify whether we should put GitHub usersnames, GitLab usernames, email addresses or real names in there.

#### Other checks
I didn't see documentation on what things cpplint actually checks, maybe we need to read [the source code](https://github.com/google/styleguide/blob/gh-pages/cpplint/cpplint.py). Several other things in there seem like a moderate amount of work.

### `ament_uncrustify`
This had no output, because I think I already ran it on the original code.

[... 10 lines cut ...]

### `clang-tidy`
The `ament_clang_tidy` package seems to do nothing, but it can be used manually:
* Build with `--cmake-args -DCMAKE_EXPORT_COMPILE_COMMANDS=1`
* In the root directory, execute `clang-tidy <cpp file paths> -p build/compile_commands.json`

It is sometimes slow, but can be extremely valuable. In `pose_initializer` it found nothing, but that is a tiny package. In `surround_obstacle_checker` however it pointed out that in `rclcpp::Duration(0.5)`, 0.5 is rounded to 0 – it accepts nanoseconds, not seconds, so this was a bug. It also pointed out an unnecessary copy, an unused argument to a function, and the use of potentially-moved-from objects.

[... cut ...] -apex-ai can comment more on it since he has used it in the past.

My suggestion is to try to batch-migrate license comments, header filenames, and `#include` style. We should probably also track all `TODO` comments in an issue to add explanations and assignees. Then we can go through all packages and fix lint issues and run `clang-tidy`. And of course, there is also tests, a good overall design, build quality, and documentation etc. that we should try to improve as part of code quality efforts.

closed 2020-12-14 by mitsudome-r

fred-apex-ai, 2020-12-14:
> Requiring header guards instead of #pragma once
> I didn't find a tool to generate those automatically, so it would potentially mean a moderate amount of work for very little gain, at least where #pragma once is already there.

@nnmm There is a good tool that I have used for this purpose in the past and on at least one package during porting until I decided this should be done in a batch operation later. It's https://github.com/cgmb/guardonce. THe workflow would be to first convert everything to `#pragma once` to have a common ground, then convert to include guards based on the path in the repo. But we can/should do that in autoware.auto when the position in the directory structure would change again. Pseudocode

```
for each header:
    guard2once header
   once2guard header -p "format based on file name"
```