tier4/AutowareArchitectureProposal.proj #155: Error in building system monitor

ayberksener, 2020-12-24:
Hi, I got the following error upon building the traffic_light_classifier
 error: ‘nvmlClocksThrottleReasonDisplayClockSetting’ was not declared in this scope
                                  : ((X)&nvmlClocksThrottleReasonDisplayClockSetting) 

Kindly help me in this regard.

taichiH, 2021-02-08:
Could you please share me your developing environment?

LalithVipulananthan, 2021-04-16:
@ayberksener Hi there. Could you confirm if this issue is still occuring with the latest v0.9.1 branch code please?

LalithVipulananthan, 2021-05-07:
Closing issue due to lack of response from OP. @ayberksener If this issue is still occuring, then please provide more information about your development environment so that we can assist further.

closed 2021-05-07 by LalithVipulananthan