tier4/AutowareArchitectureProposal.proj #173: what is different sensor model 1)aip_s1 2) aip_x1 3) aip_xx2

lejk8104, 2021-02-22:
`roslaunch autoware_launch logging_simulator.launch map_path:=[path] vehicle_model:=lexus sensor_model:=aip_xx1 rosbag:=true`
default sensor model: aip_xx1 is support 1)(top/rear/left/right) lidar 2) gnss 3) camera 4)imu
but I don't exactly different other sensors config
my question is 
1) what is mainly different aip_s1, aip_x1, aip_x2
2) what is sensor_description/aip_xx1/sensor_kit.xacro? i want to know this code's purpose and operation work.
3) I think another sensor model is not supported "sensor_kit_calibration" is right? because other sensor_kit_catlibration file is empty!!

1) link "sensor_kit.xacro" : https://drive.google.com/file/d/19WZ6QC8La2hr1CWoOz-ZgzsOQq4oVCmX/view?usp=sharing
2) link "sensor_xx1-sensor_kit_calibration : https://drive.google.com/file/d/13ofO9E3W6m7gFaLb8xgdUVu0exsK_QN7/view?usp=sharing
2) link "other sensor_kit_calibration is empty" : https://drive.google.com/file/d/13xuW4EwcLXINX_XkkRzbLjtzrEWsi4O2/view?usp=sharing

yukkysaito, 2021-05-31:
Thanks for using it.

>what is mainly different aip_s1, aip_x1, aip_x2

This is the name of the sensor kit we use in the company. And we choose the sensor kit we use based on ODD (Operational Design Domain) and cost.

>what is sensor_description/aip_xx1/sensor_kit.xacro? i want to know this code's purpose and operation work.

sensor_kit is the roofiunit on top of the taxi [here](https://www.watch.impress.co.jp/docs/news/1287297.html)
It can be separated as a sensor kit  and only the mounting position can be managed.

>I think another sensor model is not supported "sensor_kit_calibration" is right? because other sensor_kit_catlibration file is empty!!

Yes. You are right.
In the public version, all sensor kits are not published and are undefined

kenji-miyake, 2021-09-15:
Thank you for your contributions/questions.
I'll close this according to [our contribution guidelines](https://github.com/tier4/AutowareArchitectureProposal.proj/blob/main/CONTRIBUTING.md).

closed 2021-09-15 by kenji-miyake