tier4/AutowareArchitectureProposal.proj #200: Maptool

likui01, 2021-05-03:
Hi, 
I am using an online maptool to make lanelet vector map of my univeristy. 
i have a PCD 3D map of the university. When I draw the lanelet using the tool. I get the error that "next lanelet is empty"
As shown in the image. can you help me in this regard
thank you 
[image]

LalithVipulananthan, 2021-05-07:
@likui01 Hello there! I work at Tier IV and I can try and help out with this issue. Is the following a correct statement of the steps that you took?
1. Loaded an existing PCD map into the Vector Map Builder
2. Created a Lanelet2 vector map
3. Tried adding a Lanelet and then the "next lanelet is empty" error occurred

Does this happen whenever and wherever on the map you try to add a Lanelet? Would you be able to share your PCD map data with me so that I can take a closer look to try to figure out the problem?

likui01, 2021-05-17:
@LalithVipulananthan 
Thank you for your response. 
I want to ask, how can I make a lane with dash lane marking. 
I used the centerline tool to create dash lane marking as shown in the figure below.
[image]
But when I used it with autoware, the autoware took the centerline as waypoints and car travel on them.
[image]
How can make multi lane? 
thank you

LalithVipulananthan, 2021-05-25:
@likui01 Apologies for not replying sooner. I answered your first question as a courtesy, but as it seems that you have additional questions that are unrelated to the AutowareArchitectureProposal repository, I think it would be better to ask this question (and any others that you may have) on the ROS Answers site. Thank you for using our Vector Map Builder product, and I hope that your issue can be resolved quickly.

closed 2021-05-25 by LalithVipulananthan