tier4/AutowareArchitectureProposal.proj #206: ./setup_ubuntu18.04.sh error

1129267324, 2021-08-04:
Setting up the pre-build environment for AutowareArchitectureProposal.IV can take up to 45 minutes.
>  Are you sure you want to run setup? [y/n] y
SUDO password: 
 [WARNING]: Unable to parse
/home/sdas/workspace/AutowareArchitectureProposal.proj/inventories/local-
dev.ini as an inventory source

Would you like to install the optional GPU modules (cuda: 10.2, cudnn: 7.6.5, TensorRT: 7.0.0)? (y/n) [y]: ^C [ERROR]: User interrupted execution

Complete 
sdas@sdas-Default-string:~/workspace/AutowareArchitectureProposal.proj$ ./setup_ubuntu18.04.sh 
Setting up the pre-build environment for AutowareArchitectureProposal.IV can take up to 45 minutes.
>  Are you sure you want to run setup? [y/n] y
SUDO password: 
 [WARNING]: Unable to parse
/home/sdas/workspace/AutowareArchitectureProposal.proj/inventories/local-
dev.ini as an inventory source

Would you like to install the optional GPU modules (cuda: 10.2, cudnn: 7.6.5, TensorRT: 7.0.0)? (y/n) [y]: n

PLAY [localhost] ***************************************************************

TASK [Gathering Facts] *********************************************************
ok: [localhost]

TASK [pause] *******************************************************************
Pausing for 10 seconds
(ctrl+C then 'C' = continue early, ctrl+C then 'A' = abort)
[pause]
[Warning]: Skipping installation of GPU modules. To execute nodes depending on TensorRT, please manually install the modules listed above.:
ok: [localhost]

TASK [cuda : CUDA (add nvidia key)] ********************************************
skipping: [localhost]

TASK [cuda : CUDA (install CUDA 10.2 repository)] ******************************
skipping: [localhost]

TASK [cuda : CUDA (install CUDA 10.2)] *****************************************
skipping: [localhost]

TASK [cuda : CUDA (add path >> bashrc)] ****************************************
skipping: [localhost]

TASK [cuda : CUDA (add library >> bashrc)] *************************************
skipping: [localhost]

TASK [tensorrt : TensorRT (install machine learning repository)] ***************
skipping: [localhost]

TASK [tensorrt : TensorRT (install libraries for TensorRT 7)] ******************
skipping: [localhost]

TASK [tensorrt : TensorRT (prevent cuda related packages from upgrading)] ******
skipping: [localhost] => (item=cuda-repo-ubuntu1804) 
skipping: [localhost] => (item=nvidia-machine-learning-repo-ubuntu1804) 
skipping: [localhost] => (item=cuda-10-2) 
skipping: [localhost] => (item=libcudnn7) 
skipping: [localhost] => (item=libcudnn7-dev) 
skipping: [localhost] => (item=libnvinfer7) 
skipping: [localhost] => (item=libnvinfer-dev) 
skipping: [localhost] => (item=libnvinfer-plugin7) 
skipping: [localhost] => (item=libnvinfer-plugin-dev) 
skipping: [localhost] => (item=libnvonnxparsers7) 
skipping: [localhost] => (item=libnvonnxparsers-dev) 
skipping: [localhost] => (item=libnvparsers7) 
skipping: [localhost] => (item=libnvparsers-dev) 

TASK [ros : ROS (set up your keys)] ********************************************
ok: [localhost]

TASK [ros : ROS (add source list)] *********************************************
ok: [localhost]

TASK [ros : ROS (install packages)] ********************************************
ok: [localhost]

TASK [ros : ROS (check rosdep file)] *******************************************
ok: [localhost]

TASK [ros : ROS (install rosdep)] **********************************************
ok: [localhost]

TASK [ros : ROS (initialize rosdep)] *******************************************
skipping: [localhost]

TASK [ros : ROS (update rosdep)] ***********************************************
changed: [localhost]

TASK [ros : ROS (add path >> bashrc)] ******************************************
ok: [localhost]

TASK [ros : ROS (install dependencies for building packages)] ******************
ok: [localhost]

TASK [osqp : osqp (clone repository)] ******************************************
ok: [localhost]

TASK [osqp : osqp (make build dir)] ********************************************
ok: [localhost] => (item=build)

TASK [osqp : osqp (cmake)] *****************************************************
changed: [localhost]

TASK [osqp : osqp (make all and install)] **************************************
changed: [localhost]

TASK [autoware : Autoware (update rosdep)] *************************************
fatal: [localhost]: FAILED! => {"changed": true, "cmd": ["rosdep", "update"], "delta": "0:00:00.133947", "end": "2021-08-04 17:14:18.668351", "msg": "non-zero return code", "rc": 1, "start": "2021-08-04 17:14:18.534404", "stderr": "Traceback (most recent call last):\n  File \"/usr/local/bin/rosdep\", line 5, in <module>\n    from rosdep2.main import rosdep_main\n  File \"/usr/local/lib/python3.6/dist-packages/rosdep2/__init__.py\", line 45, in <module>\n    from .lookup import RosdepDefinition, RosdepView, RosdepLookup, \\\n  File \"/usr/local/lib/python3.6/dist-packages/rosdep2/lookup.py\", line 44, in <module>\n    from .sources_list import SourcesListLoader\n  File \"/usr/local/lib/python3.6/dist-packages/rosdep2/sources_list.py\", line 50, in <module>\n    from .gbpdistro_support import get_gbprepo_as_rosdep_data, download_gbpdistro_as_rosdep_data\n  File \"/usr/local/lib/python3.6/dist-packages/rosdep2/gbpdistro_support.py\", line 18, in <module>\n    from .platforms.debian import APT_INSTALLER\n  File \"/usr/local/lib/python3.6/dist-packages/rosdep2/platforms/debian.py\", line 34, in <module>\n    from rospkg.os_detect import (\nImportError: cannot import name 'OS_POP'", "stderr_lines": ["Traceback (most recent call last):", "  File \"/usr/local/bin/rosdep\", line 5, in <module>", "    from rosdep2.main import rosdep_main", "  File \"/usr/local/lib/python3.6/dist-packages/rosdep2/__init__.py\", line 45, in <module>", "    from .lookup import RosdepDefinition, RosdepView, RosdepLookup, \\", "  File \"/usr/local/lib/python3.6/dist-packages/rosdep2/lookup.py\", line 44, in <module>", "    from .sources_list import SourcesListLoader", "  File \"/usr/local/lib/python3.6/dist-packages/rosdep2/sources_list.py\", line 50, in <module>", "    from .gbpdistro_support import get_gbprepo_as_rosdep_data, download_gbpdistro_as_rosdep_data", "  File \"/usr/local/lib/python3.6/dist-packages/rosdep2/gbpdistro_support.py\", line 18, in <module>", "    from .platforms.debian import APT_INSTALLER", "  File \"/usr/local/lib/python3.6/dist-packages/rosdep2/platforms/debian.py\", line 34, in <module>", "    from rospkg.os_detect import (", "ImportError: cannot import name 'OS_POP'"], "stdout": "", "stdout_lines": []}
	to retry, use: --limit @/home/sdas/workspace/AutowareArchitectureProposal.proj/ansible/localhost-setup-ubuntu18.04-devpc.retry

PLAY RECAP *********************************************************************
localhost                  : ok=14   changed=3    unreachable=0    failed=1   

Complete

1129267324, 2021-08-04:
how can I solve the problem, I have instlled the ros,tensorRT7.0.0,CUDA,cuDNN. But while runing colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release --catkin-skip-building-tests still have some erroes.

kenji-miyake, 2021-09-15:
Thank you for your contributions/questions.
I'll close this according to [our contribution guidelines](https://github.com/tier4/AutowareArchitectureProposal.proj/blob/main/CONTRIBUTING.md).

closed 2021-09-15 by kenji-miyake