tier4/AutowareArchitectureProposal.proj #226: terminate called after throwing an instance of 'boost::process::process_error'

jianbuzhang, 2021-11-01:
When I run this command:roslaunch autoware_launch logging_simulator.launch map_path:=/home/jinjian/WorkSpace/ws_autowareArchitectureProposal/rosbag_sample_map vehicle_model:=lexus sensor_model:=aip_xx1 rosbag:=true, I met the above mistake.
`[system/system_monitor/ntp_monitor-7] restarting process
process[system/system_monitor/ntp_monitor-7]: started with pid [16686]
terminate called after throwing an instance of 'boost::process::process_error'
  what():  execve failed: No such file or directory`

jianbuzhang, 2021-11-01:
@dejanpan @esteve @netmarkjp

kenji-miyake, 2021-11-01:
Thank you for your question.
Please read [our contribution guidelines](https://github.com/tier4/AutowareArchitectureProposal.proj/blob/main/CONTRIBUTING.md).

As far as I know, probably you haven't set up system_monitor properly, but I think you can ignore the error.

closed 2021-11-01 by kenji-miyake