tier4/AutowareArchitectureProposal.proj #249: ./setup_ubuntu20.04.sh fail to add CUDA repository into sources.list

aminballoon, 2022-08-02:
> ./setup_ubuntu20.04.sh
Setting up the build environment take up to 45 minutes.
>  Are you sure you want to run setup? [y/N] y
BECOME password: 
Would you like to install the optional GPU modules (cuda: 11.1, cudnn: 8.0.5, TensorRT: 7.2.1)? (y/n) [y]: y

PLAY [localhost] ***************************************************************

TASK [Gathering Facts] *********************************************************
ok: [localhost]

TASK [pause] *******************************************************************
skipping: [localhost]

TASK [cuda : install software-properties-common] *******************************
ok: [localhost]

TASK [cuda : CUDA (download pin file)] *****************************************
ok: [localhost]

TASK [cuda : CUDA (add nvidia key)] ********************************************
ok: [localhost]

TASK [cuda : CUDA (add CUDA repository into sources.list)] *********************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "apt cache update failed"}

PLAY RECAP *********************************************************************
localhost                  : ok=4    changed=0    unreachable=0    failed=1    skipped=1    rescued=0    ignored=0   

###########################
How can i solve this problem ??

kenji-miyake, 2022-08-02:
@aminballoon Hi, would it be possible for you to use https://github.com/autowarefoundation/autoware instead?
This repository is not maintained and will be archived soon.

aminballoon, 2022-08-02:
> @aminballoon Hi, would it be possible for you to use https://github.com/autowarefoundation/autoware instead? This repository is not maintained and will be archived soon.

Thanks for your answer, It's work !!!

closed 2022-08-02 by aminballoon